martin-ger/esp32_nat_router #3: can it change MAC address

iansun2, 2020-03-04:
can it change MAC address like [https://github.com/martin-ger/esp_wifi_repeater](url) 
using " set [ap_mac|sta_mac] xx:xx:xx:xx:xx:xx " command

martin-ger, 2020-03-04:
No, not yet, however should not be hard to add, if required using [esp_wifi_set_mac()](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_wifi.html#_CPPv416esp_wifi_set_mac16wifi_interface_tAL6E_K7uint8_t).

iansun2, 2020-03-08:
> No, not yet, however should not be hard to add, if required using [esp_wifi_set_mac()](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/network/esp_wifi.html#_CPPv416esp_wifi_set_mac16wifi_interface_tAL6E_K7uint8_t).
I cannot install esp-idf .....
I have try on windows10 and ubuntu
But I always get error 1 or 2 depend on python 2 or 3

martin-ger, 2020-03-09:
Sorry, for me on Ubuntu 18 it worked as described. Please ask in the esp-idf repository.

closed 2020-03-18 by iansun2