martin-ger/esp32_nat_router #6: build failed

JoergSH, 2020-03-16:
I Setup ESP-IDF and try to build...

> ../main/esp32_nat_router.c:110:27: error: 'UART_SCLK_REF_TICK' undeclared (first
 use in this function); did you mean 'UART_CLK_EN_M'?
             .source_clk = UART_SCLK_REF_TICK,
                           ^~~~~~~~~~~~~~~~~~
                           UART_CLK_EN_M
../main/esp32_nat_router.c:110:27: note: each undeclared identifier is reported
only once for each function it appears in
[819/827] Building C object esp-idf/cm.../__idf_cmd_system.dir/cmd_system.c.obj
FAILED: esp-idf/cmd_system/CMakeFiles/__idf_cmd_system.dir/cmd_system.c.obj
C:\Users\Joerg\.espressif\.espressif\tools\xtensa-esp32-elf\esp-2019r2-8.2.0\xte
nsa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe  -Iconfig -I../components/cmd_system
-IC:/Users/Joerg/Desktop/esp-idf/components/newlib/platform_include -IC:/Users/J
oerg/Desktop/esp-idf/components/freertos/include -IC:/Users/Joerg/Desktop/esp-id
f/components/heap/include -IC:/Users/Joerg/Desktop/esp-idf/components/log/includ
e -IC:/Users/Joerg/Desktop/esp-idf/components/soc/esp32/include -IC:/Users/Joerg
/Desktop/esp-idf/components/soc/include -IC:/Users/Joerg/Desktop/esp-idf/compone
nts/esp_rom/include -IC:/Users/Joerg/Desktop/esp-idf/components/esp_common/inclu
de -IC:/Users/Joerg/Desktop/esp-idf/components/xtensa/include -IC:/Users/Joerg/D
esktop/esp-idf/components/xtensa/esp32/include -IC:/Users/Joerg/Desktop/esp-idf/
components/esp32/include -IC:/Users/Joerg/Desktop/esp-idf/components/driver/incl
ude -IC:/Users/Joerg/Desktop/esp-idf/components/esp_ringbuf/include -IC:/Users/J
oerg/Desktop/esp-idf/components/esp_event/include -IC:/Users/Joerg/Desktop/esp-i
df/components/tcpip_adapter/include -IC:/Users/Joerg/Desktop/esp-idf/components/
lwip/include/apps -IC:/Users/Joerg/Desktop/esp-idf/components/lwip/include/apps/
sntp -IC:/Users/Joerg/Desktop/esp-idf/components/lwip/lwip/src/include -IC:/User
s/Joerg/Desktop/esp-idf/components/lwip/port/esp32/include -IC:/Users/Joerg/Desk
top/esp-idf/components/lwip/port/esp32/include/arch -IC:/Users/Joerg/Desktop/esp
-idf/components/vfs/include -IC:/Users/Joerg/Desktop/esp-idf/components/esp_wifi
/include -IC:/Users/Joerg/Desktop/esp-idf/components/esp_wifi/esp32/include -IC:
/Users/Joerg/Desktop/esp-idf/components/esp_eth/include -IC:/Users/Joerg/Desktop
/esp-idf/components/efuse/include -IC:/Users/Joerg/Desktop/esp-idf/components/ef
use/esp32/include -IC:/Users/Joerg/Desktop/esp-idf/components/app_trace/include
-IC:/Users/Joerg/Desktop/esp-idf/components/console -IC:/Users/Joerg/Desktop/esp
-idf/components/spi_flash/include -mlongcalls -Wno-frame-address   -ffunction-se
ctions -fdata-sections -fstrict-volatile-bitfields -nostdlib -Wall -Werror=all -
Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-v
ariable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-si
gn-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_V
ER=\"v4.0-dirty\" -DGCC_NOT_5_2_0 -DESP_PLATFORM -MD -MT esp-idf/cmd_system/CMak
eFiles/__idf_cmd_system.dir/cmd_system.c.obj -MF esp-idf\cmd_system\CMakeFiles\_
_idf_cmd_system.dir\cmd_system.c.obj.d -o esp-idf/cmd_system/CMakeFiles/__idf_cm
d_system.dir/cmd_system.c.obj   -c ../components/cmd_system/cmd_system.c
../components/cmd_system/cmd_system.c: In function 'light_sleep':
../components/cmd_system/cmd_system.c:294:5: error: implicit declaration of func
tion 'uart_wait_tx_idle_polling'; did you mean 'uart_wait_tx_done'? [-Werror=imp
licit-function-declaration]
     uart_wait_tx_idle_polling(CONFIG_ESP_CONSOLE_UART_NUM);
     ^~~~~~~~~~~~~~~~~~~~~~~~~
     uart_wait_tx_done
cc1.exe: some warnings being treated as errors
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\Users\Joerg\Documents\GitHub\esp32_nat_router>

iceychris, 2020-03-31:
@martin-ger @JoergSH I managed to get it to build with the following steps:

* use `esp-idf` version `v4.0`
* use `toolchain` version `esp-2019r2` (I compiled from source)
* (checkout this repo at commit `ad5e2267f79664956a26e549b9ccf7813b46360b`)
* replace lwip as explained in the `README`
* apply this diff:

```
diff --git a/components/cmd_system/cmd_system.c b/components/cmd_system/cmd_system.c
index e78e41b..3565a26 100644
--- a/components/cmd_system/cmd_system.c
+++ b/components/cmd_system/cmd_system.c
@@ -291,7 +291,7 @@ static int light_sleep(int argc, char **argv)
         ESP_ERROR_CHECK( esp_sleep_enable_uart_wakeup(CONFIG_ESP_CONSOLE_UART_NUM) );
     }
     fflush(stdout);
-    uart_wait_tx_idle_polling(CONFIG_ESP_CONSOLE_UART_NUM);
+    uart_wait_tx_done(CONFIG_ESP_CONSOLE_UART_NUM, 1);
     esp_light_sleep_start();
     esp_sleep_wakeup_cause_t cause = esp_sleep_get_wakeup_cause();
     const char *cause_str;
diff --git a/main/esp32_nat_router.c b/main/esp32_nat_router.c
index 10ae6b0..8cd527d 100644
--- a/main/esp32_nat_router.c
+++ b/main/esp32_nat_router.c
@@ -107,7 +107,6 @@ static void initialize_console(void)
             .data_bits = UART_DATA_8_BITS,
             .parity = UART_PARITY_DISABLE,
             .stop_bits = UART_STOP_BITS_1,
-            .source_clk = UART_SCLK_REF_TICK,
     };
     /* Install UART driver for interrupt-driven reads and writes */
     ESP_ERROR_CHECK( uart_driver_install(CONFIG_ESP_CONSOLE_UART_NUM,
diff --git a/main/http_server.c b/main/http_server.c
index 5d3e9c9..7324f42 100644
--- a/main/http_server.c
+++ b/main/http_server.c
@@ -14,7 +14,6 @@
 #include <esp_timer.h>
 #include <sys/param.h>
 //#include "nvs_flash.h"
-#include "esp_netif.h"
 //#include "esp_eth.h"
 //#include "protocol_examples_common.h"
```

JoergSH, 2020-03-31:
I set up a Linux to build all.
There is no problem to build and flash

Thanks Chris

closed 2020-03-31 by JoergSH