martin-ger/esp32_nat_router #7: Failed more than 2 NAT Router.

beyonlo, 2020-03-23:
Hi, I created 3 ESP32 NAT router and not works, just max 2.

**This not works:**
SmartPhone --> ESP32-NAT-03 --> ESP32-NAT-02 --> ESP32-NAT-01 --> REAL-WIFI-ROUTER --> INTERNET
Problem: SmartPhone connect to ESP32-NAT-03 but internet NOT WORKS. Just change SmartPhone to connect to ESP32-NAT-02 and everything works. If I change order between 03 and 02, so 02 not works anymore, like this: ESP32-NAT-02 --> ESP32-NAT-03 --> ESP32-NAT-01
So I think there are a limitation about more than 2 NAT nodes, is it correct?

**This works fine:**
SmartPhone --> ESP32-NAT-02 --> ESP32-NAT-01 --> REAL-WIFI-ROUTER --> INTERNET

Any ideia?

Thanks

beyonlo, 2020-04-22:
any idea?

cloudykeyru, 2020-05-28:
Change the IP addresses so that they are different on the ESP.

marfanwicaksono, 2020-09-24:
In `wifi_init` function, change the following line to different IP address for every devices:
```
    IP4_ADDR(&ipInfo.ip, 192,168,4,1); // change this IP
    IP4_ADDR(&ipInfo.gw, 192,168,4,1); // change this IP
    IP4_ADDR(&ipInfo.netmask, 255,255,255,0);
```

In `app_main` function, change the following line to different IP address for every devices:
```
#if IP_NAPT
    u32_t napt_netif_ip = 0xC0A80401; // Set to ip address of softAP netif (Default is 192.168.4.1) (change this IP using hex representation)
    ip_napt_enable(htonl(napt_netif_ip), 1);
    ESP_LOGI(TAG, "NAT is enabled");
#endif
```

I've tried it using four different devices with IP addresses 192.168.1.1, 192.168.2.1, 192.168.3.1, 192.168.4.1, respectively and its working.

beyonlo, 2020-09-24:
> I've tried it using four different devices with IP addresses 192.168.1.1, 192.168.2.1, 192.168.3.1, 192.168.4.1, respectively and its working.

@marfanwicaksono 

Great solution. Problem with that is to need to generate many fimrwares, I mean, one firmware for each IP address that I want. 
Maybe a better solution is to have on the web interface a place to user choose the IP for each device (AP).

Thanks.

beta-tester, 2021-02-16:
via:

- web interface + GPIO to reset to factory defaults
- and/or GPIO jumper selecting IP out of two or four predefined IPs
- and/or serial port command to set IP

ArhiLeeve, 2021-02-19:
Is there away to add custom mac address to wifiSTA.

beta-tester, 2021-02-22:
@ArhiLeeve maybe you can add a method in the main entrypoint to customize the base mac just when ESP32 boots.
see https://github.com/espressif/esp-idf/blob/master/docs/en/api-reference/system/system.rst#base-mac-address
this woiuld be maybe the easiest way, without looking for the point where you can kick in to change the wifiSTA.mac.

risinek, 2021-04-10:
@ArhiLeeve yes, there is a function for this in `esp_wifi` component [`esp_wifi_set_mac()`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/network/esp_wifi.html#_CPPv416esp_wifi_set_mac16wifi_interface_tAL6E_K7uint8_t)