martin-ger/esp32_nat_router #10: What's the maxmum client number please

hotsnow, 2020-04-07:
For esp8266 max client numbers are 8 due to the ESP's limit. How many clients can connect simultaneously for esp32? I have lots of IoT devices need to connect to it. I just want to know if there is a limit too. Bandwidth is not a problem for these devices, but 8 client is too small for me :( Thanks.

martin-ger, 2020-04-07:
According to /components/esp_wifi/include/esp_wifi_types.h in the ESP-IDF the ESP_WIFI_MAX_CONN_NUM is 10. This cannot be increased (without the sources of the WiFi driver which is closed).

hotsnow, 2020-04-08:
That's not enough for me. Thank you!
PS: this is a really cool project. I love it.

beyonlo, 2020-04-09:
@hotsnow You can to put more than one ESP32 as Router working together. Ex: 4 ESP32 can to connect 36 clients IoT devices.

hotsnow, 2020-04-19:
@beyonlo then a second-hand cheap router would be a better option :)

closed 2020-04-19 by hotsnow

Marc-Dumagpi, 2024-11-24:
Good Afternoon, Can i make the max users is just 1, how do i do this or where can i find this part of the code

nissimzur, 2024-11-24:
�

Hi
I believe you have to make an array with USER names and passwords. 
when you get USER name request, you can find the corresponded password and use it.

 

בכבוד רב,

ד"ר נסים צור

***@***.***

Phone +972544500635
Whatsapp:  <https://wa.me/+972544500635> https://wa.me/+972544500635