martin-ger/esp32_nat_router #11: Build failed

dhirajshrotri, 2020-04-09:
I was trying to build this project on Linux Mint 19 and followed all the steps in the ReadMe. The build is failing with the following message: 

`idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/dhiraj/Workplace/esp32_nat_router/build
Executing "ninja all"...
[1/461] cd /home/dhiraj/Workplace/esp32_nat_router/build/esp-idf/partit...**********************************************************************"
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
storage,data,fat,0x110000,1M,
*******************************************************************************
[2/461] Performing build step for 'bootloader'
ninja: no work to do.
[12/459] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/spi_bus_lock.c.obj
FAILED: esp-idf/driver/CMakeFiles/__idf_driver.dir/spi_bus_lock.c.obj 
/home/dhiraj/.espressif/tools/xtensa-esp32-elf/esp-2020r1-8.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc  -Iconfig -I/home/dhiraj/Workplace/esp-idf/components/driver/include -I/home/dhiraj/Workplace/esp-idf/components/driver/esp32/include -I/home/dhiraj/Workplace/esp-idf/components/driver/include/driver -I/home/dhiraj/Workplace/esp-idf/components/newlib/platform_include -I/home/dhiraj/Workplace/esp-idf/components/freertos/include -I/home/dhiraj/Workplace/esp-idf/components/freertos/xtensa/include -I/home/dhiraj/Workplace/esp-idf/components/heap/include -I/home/dhiraj/Workplace/esp-idf/components/log/include -I/home/dhiraj/Workplace/esp-idf/components/lwip/include/apps -I/home/dhiraj/Workplace/esp-idf/components/lwip/include/apps/sntp -I/home/dhiraj/Workplace/esp-idf/components/lwip/lwip/src/include -I/home/dhiraj/Workplace/esp-idf/components/lwip/port/esp32/include -I/home/dhiraj/Workplace/esp-idf/components/lwip/port/esp32/include/arch -I/home/dhiraj/Workplace/esp-idf/components/soc/src/esp32/. -I/home/dhiraj/Workplace/esp-idf/components/soc/src/esp32/include -I/home/dhiraj/Workplace/esp-idf/components/soc/include -I/home/dhiraj/Workplace/esp-idf/components/esp_rom/include -I/home/dhiraj/Workplace/esp-idf/components/esp_common/include -I/home/dhiraj/Workplace/esp-idf/components/esp_system/include -I/home/dhiraj/Workplace/esp-idf/components/xtensa/include -I/home/dhiraj/Workplace/esp-idf/components/xtensa/esp32/include -I/home/dhiraj/Workplace/esp-idf/components/esp32/include -I/home/dhiraj/Workplace/esp-idf/components/esp_ringbuf/include -I/home/dhiraj/Workplace/esp-idf/components/efuse/include -I/home/dhiraj/Workplace/esp-idf/components/efuse/esp32/include -I/home/dhiraj/Workplace/esp-idf/components/espcoredump/include -I/home/dhiraj/Workplace/esp-idf/components/esp_timer/include -I/home/dhiraj/Workplace/esp-idf/components/soc/soc/esp32/. -I/home/dhiraj/Workplace/esp-idf/components/soc/soc/esp32/include -I/home/dhiraj/Workplace/esp-idf/components/soc/soc/esp32/../include -I/home/dhiraj/Workplace/esp-idf/components/vfs/include -I/home/dhiraj/Workplace/esp-idf/components/esp_wifi/include -I/home/dhiraj/Workplace/esp-idf/components/esp_wifi/esp32/include -I/home/dhiraj/Workplace/esp-idf/components/esp_event/include -I/home/dhiraj/Workplace/esp-idf/components/esp_netif/include -I/home/dhiraj/Workplace/esp-idf/components/esp_eth/include -I/home/dhiraj/Workplace/esp-idf/components/tcpip_adapter/include -I/home/dhiraj/Workplace/esp-idf/components/app_trace/include -mlongcalls -Wno-frame-address   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -Wno-parentheses -Wno-sizeof-pointer-memaccess -Wno-clobbered -Wno-format-overflow -Wno-stringop-truncation -Wno-misleading-indentation -Wno-cast-function-type -Wno-implicit-fallthrough -Wno-unused-const-variable -Wno-switch-unreachable -Wno-format-truncation -Wno-memset-elt-size -Wno-int-in-bool-context -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.2-dev-1097-g2e14149bf-dirty\" -DESP_PLATFORM -MD -MT esp-idf/driver/CMakeFiles/__idf_driver.dir/spi_bus_lock.c.obj -MF esp-idf/driver/CMakeFiles/__idf_driver.dir/spi_bus_lock.c.obj.d -o esp-idf/driver/CMakeFiles/__idf_driver.dir/spi_bus_lock.c.obj   -c /home/dhiraj/Workplace/esp-idf/components/driver/spi_bus_lock.c
/home/dhiraj/Workplace/esp-idf/components/driver/spi_bus_lock.c: In function 'spi_bus_lock_init_main_dev':
/home/dhiraj/Workplace/esp-idf/components/driver/spi_bus_lock.c:822:41: error: implicit declaration of function 'xSemaphoreCreateBinaryStatic'; did you mean 'xSemaphoreCreateBinary'? [-Werror=implicit-function-declaration]
     g_spi_lock_main_flash_dev->semphr = xSemaphoreCreateBinaryStatic(&main_flash_semphr);
                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                         xSemaphoreCreateBinary
/home/dhiraj/Workplace/esp-idf/components/driver/spi_bus_lock.c:822:39: warning: assignment to 'SemaphoreHandle_t' {aka 'void *'} from 'int' makes pointer from integer without a cast [-Wint-conversion]
     g_spi_lock_main_flash_dev->semphr = xSemaphoreCreateBinaryStatic(&main_flash_semphr);
                                       ^
cc1: some warnings being treated as errors
[21/459] Building C object esp-idf/esp32/CMakeFiles/__idf_esp32.dir/sleep_modes.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

Any ideas as to what maybe causing this?

wuppergit, 2020-04-21:
Hi Martin,
very great work. The included binaries are working fine. 
But I'm not able to build them on my System. I've tried a lot:
Windows 10:
esp-idf V3.2: cmake error
            V4  : esp_netif.h missing
            master:  error as above
Ubuntu 18.10:
            V4  : esp_netif.h missing
            master:  error as above
 Build-Command: idf.py build
Which environment / build-command did you use?
Any idea to get on?
Thanks

wuppergit, 2020-04-22:
ESP-IDF v4.2-dev-1126-gd85d3d969-dirty
It's a bug in components\freertos\include\freertos\semphr.h (look at [https://github.com/espressif/esp-idf/issues/5046](url) )
workaround: menucofig -> Component config -> FreeRTOS ->  Enable FreeRTOS static allocation API
Build-process successful with waringns

bridystone, 2020-04-22:
Hi,

i also ran in the same issue recently - trying to build the project.
another option to solve it is to use the stable "release/v4.1" of  the ESP-IDF.

I *didn't* enable the FreeRTOS static allocation API - and it worked as well.

(Finally after lots of trouble with my python on macOS High Sierra...)

THANKS to Martin for the project!
Jens

Xiaoxiao1990, 2020-05-20:
I build the project with ESP-IDF v4.0 and v4.2-dev, it required lwip/src/core/ipv4/ip4_nat.h，but it doesn't exist in your esp-lwip repository. May I ask for your build ESP-IDF version?

Thanks a lot.

dhirajshrotri, 2020-05-20:
First of all sorry for not commenting on this issue for so long.

> I build the project with ESP-IDF v4.0 and v4.2-dev, it required lwip/src/core/ipv4/ip4_nat.h，but it doesn't exist in your esp-lwip repository. May I ask for your build ESP-IDF version?
> 
> Thanks a lot.

I am using the v4.2-dev

> I thnk espressif has played a little bit around. The source-file spi_bus_lock.c is definitly wrong and it is not done to change xSemaphoreCreateBinaryStatic to xSemaphoreCreateBinary. The file is new at master branche since 18 days.
> All in all espressf is atually not very professional.

Changing xSemaphoreCreateBinaryStatic to xSemaphoreCreateBinary solved the issue for me. Thanks a lot to @wuppergit !

Can I close this issue?

Xiaoxiao1990, 2020-05-20:
> First of all sorry for not commenting on this issue for so long.
>
>
> I am using the v4.2-dev
>
>
> Changing xSemaphoreCreateBinaryStatic to xSemaphoreCreateBinary solved the issue for me. Thanks a lot to [...]

I am build it successful with v4.2-dev, thanks for your nice reply. And it doesn't need replace the `components/lwip/lwip` with `esp-lwip` which described in README. 
I am config ON the SDK: 
1. `Component config > LWIP > [*] Enable copy between Layer2 and Layer3 packets`
2. `Component config > LWIP > [*] Enable IP forwarding`
3. `Component config > LWIP > [*]         Enable NAT (new/experimental)`

Here, all those functions runs well.

FLASHXE, 2020-07-05:
> I am build it successful with v4.2-dev, thanks for your nice reply. And it doesn't need replace the `components/lwip/lwip` with `esp-lwip` which described in README.
> I am config ON the SDK:
>
>     1. [...]

Thank you very much. I have been troubled by this problem for a long time. After downloading the tool and IDF4.2 with difficulty, it is ready to use

FLASHXE, 2020-07-05:
> I am build it successful with v4.2-dev, thanks for your nice reply. And it doesn't need replace the `components/lwip/lwip` with `esp-lwip` which described in README.
> I am config ON the SDK:
>
>     1. [...]

Hi, I compiled it without replacing liWP, but it doesn't work as NAPT. And you?

dhirajshrotri, 2020-07-06:
I'm closing this issue since there are no more errors and everything is working.

closed 2020-07-06 by dhirajshrotri