martin-ger/esp32_nat_router #12: Missing an status-indicating GPIO-LED

CRprog, 2020-04-12:
At first I've tested your ESP8266-nat-router, it works ... But unfortunately it was much too slow for my Smart TV. Also the connection to AP (Fritzbox) after Powerup/Reset did not always work. (I admit: The signal strength is low at RSSI: -80)
With the ESP32-software here it works now much more reliably and fast enough at exact the same position and orientation of the ESP-Antenna.
What I'am missing compared to the ESP8266-Version is a LED-Output (maybe GPIO2) to indicate
"No AP connect"/"AP connect"/"AP connect with traffic".
Maybe you will find some time to integrate that later.
Regards.

on2aav, 2020-05-17:
Same here, all works greatly, just missing the GPIO2 Led when connected, or when activity.

CRprog, 2021-04-08:
Thank you for adding the Status-LED-feature last month, but I haven't tested it yet.
Nevertheless, I think I can close this issue now.

closed 2021-04-08 by CRprog