martin-ger/esp32_nat_router #14: Arduino compatibility

rw-django-fan-2020, 2020-04-15:
Is there a way to integrate the NAT feature into a Arduino project?

martin-ger, 2020-04-15:
Since release 2.6.0 NAPT/NAT is part of the standard distribution. See also this example: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/RangeExtender-NAPT/RangeExtender-NAPT.ino

zekageri, 2020-09-28:
oh god how good it would be

martin-ger, 2020-09-28:
Guess you can do that using Arduino as a component of the ESP-IDF: https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md
In the IDF menuconfig use these setting also:
https://github.com/martin-ger/esp32_nat_router#building-the-binaries

zekageri, 2020-09-29:
I'm gonna check on this! Thanks.

paclema, 2021-04-16:
Do you have some hints on how to integrate that like in the esp8266 release 2.6.0? I want to add this functionality to some other project that it is using platformio, but I am not sure how to compile your code and make it accessible to platformio exactly.

Thank you!

martin-ger, 2021-04-17:
The NAT feature is actually included in ESP8266 Arduino, have a look here:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/RangeExtender-NAPT/RangeExtender-NAPT.ino

paclema, 2021-04-17:
Thank you @martin-ger ! This is exactly what I want to approach but for the ESP32, not for the ESP8266, but for the ESP32 the lwip libraries are not available like in this example and I am not sure how to integrate them.

Do you know how could I make your awesome libraries accessible to Arduino core for the ESP32?

martin-ger, 2021-04-17:
Okay, I understand. The later ESP32 Arduino core includes the lwip_napt.h from the esp-idf ( https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/lwip/lwip/src/include/lwip/lwip_napt.h ).

So I guess it is included in liblwip and you can just include and call it. Haven't tried this so far.

If not, the libs are built without this option on. Then I would try to build the lib with esp-idf an copy it over.

paclema, 2021-04-27:
Thank you @martin-ger for your tips! As you mention, I had to re-build esp-idf libs after enabling IP forwarding and NAT so I can access to lwip_napt.h added in the latest ESP32 Arduino core. 

I have created this example [esp32_lwip_nat_example](https://github.com/paclema/esp32_lwip_nat_example) using my fork for arduino-esp32 framework where I have enabled the LWIP lib. It compiles successfully, however, it is still missing the last part to provide DNS to dhcp server. It looks that the core also contains the [dhcpserver](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/lwip/include/apps/dhcpserver/dhcpserver.h) however if I try to call the `dhcps_set_option_info()` and `dhcps_dns_setserver()` functions, the compiler fails to link the libraries giving the errors:
````
undefined 
reference to `dhcps_set_option_info(unsigned char, void*, unsigned int)'
undefined 
reference to `dhcps_dns_setserver(ip_addr const*)'
````

With this example, I can connect the esp32 as a station and I can also connect devices to it, but they don't get internet connection.

martin-ger, 2021-04-27:
Does it change anything if you include "dhcpserver/dhcpserver.h" and "dhcpserver/dhcpserver_options.h" as "extern "C" { ... }?

paclema, 2021-04-27:
I think that the libraries are included correctly directly [here](https://github.com/paclema/esp32_lwip_nat_example/blob/main/src/main.cpp#L26) but I tried adding:

```
extern "C" {
  #include "dhcpserver/dhcpserver.h"
  #include "dhcpserver/dhcpserver_options.h"
}
```
And it also includes the libraries correctly but if I uncomment the functions `dhcps_set_option_info` and `dhcps_dns_setserver` I still get the linking errors.

I am not sure if this [lwip/apps/dhcpserver/dhcpserver.c](https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/lwip/apps/dhcpserver/dhcpserver.c) is perhaps not added correctly into the sdk libraries?

paclema, 2021-05-06:
I have found the solution to accomplish the last part to provide DNS to DHCP server. The solution is described in this  [arduino-esp32 issue #5117](https://github.com/espressif/arduino-esp32/issues/5117).

You can also check the working example at this repo: [esp32_lwip_nat_example](https://github.com/paclema/esp32_lwip_nat_example)