martin-ger/esp32_nat_router #15: No documentation on port forwarding

mailmartinviljoen, 2020-04-24:
Hi there.
This is a very useful tool to have and absolutely love it!  I'm surprised how well this works by both your code and the ESP32. Thank you.

I have a pc in my garage which I use for 3d printing/CNC and showing electronic projects on screen ect ect

It want to be able to RDP to it via TCP:3389 but seeing its on a different network 192.168.4.xxx
Its not reachable from my WIFI LAN on 192.168.1.xxx. how can port forward from my network 192.168.1.xxx to 192.168.4.2 ?

Apology if this has been covered, I don't see any documentation.

cbacon93, 2021-01-22:
You want to add a route / port forward to the nat routing table.
I guess you can do this by this function: 
https://github.com/martin-ger/esp-lwip/blob/cf4da9ef1ac82efb1128aab6d0d5dd51583d56e0/src/include/lwip/lwip_napt.h#L95

tilman, 2021-03-09:
@mailmartinviljoen did you get the port forwording to work?

martin-ger, 2021-03-09:
Add the call(s) to ip_portmap_add() after esp32_nat_router.c#L367

tilman, 2021-03-10:
Thanks a lot! Worked perfectly 👍

geidien, 2023-10-24:
The code has since changed, where would you add the ip_portmap_add() calls now?