martin-ger/esp32_nat_router #16: PPPoS to WiFi Nat not working

Dude2372, 2020-04-29:
Hi,
I've built a simple application to connect my Ubuntu Laptop via PPPoS with the Esp32. The esp is connected to my Smartphone via wifi.

Pinging the esp with Ubuntu works, like Pinging the esp with Another Laptop connected to the same wifi. 
But Pinging my Smartphone by Ubuntu falls. 

The simple wifi Station - access point example works pretty fine with my Smartphone as Hotspot.

Is there an issue in this nat Implementation?

Torarqui, 2020-05-12:
I'm trying to implement this NAT with PPP and Wifi too. In my case I want the devices connected to ESP32 via Wifi to have internet access wich Is provided by a 4g modem in PPP.

Dude2372, 2020-05-15:
Did you manage to get it work?

Torarqui, 2020-05-16:
Yeah, but from what I understand of your case, wifi is the interface in which you enabled NAT, so your smartphone is behind the NAT, you can't see it from your laptop because it doesn't have an public IP, the only device you're going to see is the esp32 in between, and this is expected behaviour.

will-emmerson, 2020-08-10:
> In my case I want the devices connected to ESP32 via Wifi to have internet access wich Is provided by a 4g modem in PPP.

I'm also interested in this use case, did you manage to get it to work?

beyonlo, 2020-09-24:
> Yeah, but from what I understand of your case, wifi is the interface in which you enabled NAT, so your smartphone is behind the NAT, you can't see it from your laptop because it doesn't have an public IP, the only device you're going to see is the esp32 in between, and this is expected behaviour.

Hello @Torarqui 

Could you please public for us the solution to works to have internet access wich Is provided by a 4g modem in PPP?

Thanks.

Laiany, 2021-04-15:
I'm interested in this issue too, did anyone get it? 
Thanks

Torarqui, 2021-04-16:
@beyonlo Unfortunately I cannot share the code publicly as it is owned by the company where I work, sorry :(

beyonlo, 2021-04-16:
> @beyonlo Unfortunately I cannot share the code publicly as it is owned by the company where I work, sorry :(

Hello @Torarqui 

If you are using source code from this project to do changes, maybe (need to check license) you need to share your modifications - it is a legal notice.

@martin-ger please, could you clarify for us about what is the license type used in this project?

Thanks.