martin-ger/esp32_nat_router #17: sta connection maximum SSID: 31char

brazoayeye, 2020-05-06:
1. When set an SSID with 32 char, the sta try to connect to an SSID with first 31 entered char (last char is ignored)
2. User interface is too small to show all the 32 char in the web server, but the complete SSID is present in that field. Console print SSID with all chars

brazoayeye, 2020-06-04:
Me and my imaginary friend would appreciate a solution. 

I've been bumped today, and I forwarded. 

Thanks!

martin-ger, 2020-06-04:
To my understanding the ESP32 only supports SSIDs with max len of 31. See https://github.com/espressif/esp-idf/blob/7d75213674b4572f90c68162ad6fe9b16dae65ad/components/esp_wifi/include/esp_wifi_types.h#L225
(32 chars - NULL terminated):
`/** @brief STA configuration settings for the ESP32 */
typedef struct {
    uint8_t ssid[32];      /**< SSID of target AP. Null terminated string. */
    uint8_t password[64];  /**< Password of target AP. Null terminated string.*/`

brazoayeye, 2020-06-04:
Ok, I'll report it to the idf git. Some internet providers gives a router with a fixed 32-length password, so it's not an unusual situation....