martin-ger/esp32_nat_router #24: ESP32 cannot get internet on the 3rd node.

sagar448, 2020-07-30:
Hello @martin-ger 

We created a daisy chain with the esp32s with 3 repeaters. For some reason on the 3rd one we have no internet access at all. Why is this? How was the esp8266 able to achieve 5 nodes?

Please let us know. 
I can tell you are not active on replying to issues but we would be grateful if you could. Only since you didn't open source the library so there is no way for other people to just fix the issues so the least of your concerns should be to help people.

Thanks for understanding.
Kind Regards
Sagar

martin-ger, 2020-07-30:
What do you mean: I didn't open the sources? It is all open source! The NAPT lib expension is now even part of espressif's esp_idf (https://github.com/espressif/esp-lwip).  

The fact, that I provide binaries is just a service to all, that do not want to install the development environment. You can compile it directly from the sources provided here.

The problems in chaining might result from adressing issues: when I chained them with the ESP8266 I used a different IP subnet for each mesh level.

sagar448, 2020-07-30:
Hello @martin-ger 

Thank you for your prompt reply!
Thanks for clarifying. I understand.

Could you please just really quickly tell me how you changed the IP subnet for the mesh level in the ESP8266!
Would really appreciate it.

Thank you.
Kind Regards
Sagar

martin-ger, 2020-07-30:
In the esp_wifi_repeater the determination which mesh level a node has is a complete protocol described here: https://github.com/martin-ger/esp_wifi_repeater#automesh-mode

Is is not implemented in this software. However, you can modify the subnet here: https://github.com/martin-ger/esp32_nat_router/blob/master/main/esp32_nat_router.c#L305 and recompile it.

Maybe setting it as a config param would be the more general solution.

deostroll, 2020-08-31:
> https://github.com/martin-ger/esp32_nat_router/blob/master/main/esp32_nat_router.c#L305

Can you explain how that hex value translates to IP address?

beyonlo, 2020-09-24:
This problem as well been treated  here #7