martin-ger/esp32_nat_router #27: Header fields are too long for server to interpret

quantenschaum, 2020-09-08:
1. Flashed firmware
2. Connected to `ESP32_NAT_Router` AP
3. Setting STA Settings: SSID=foobar Password=[63 ASCII chars]
4. Click connect

The I get `Header fields are too long for server to interpret` as response.

63 chars is the max password length for WPA2. Is it possible, that the firmware cannot handle this?

korry, 2020-09-09:
I had the same problem - I closed the browser tab I had been using, opened a new tab, and browsed to 192.168.4.1 and problem solved.  Hope that helps.

lecrams, 2021-07-22:
having exactly the same issue however my PW is only 20 ASCII characters.
When I connect to the new AP i get the following logging on the UART console:

W (16264) httpd_parse: parse_block: request URI/header too long
W (16264) httpd_txrx: httpd_resp_send_err: 431 Request Header Fields Too Large - Header fields are too long for server to interpret.

Any idea what can be wrong.closing the browser does not work for me

martin-ger, 2021-07-22:
Does ist help to increase HTTPD_MAX_REQ_HDR_LEN in the configuration?

lecrams, 2021-07-22:
I changed HTTPD_MAX_REQ_HDR_LEN to 1024 rebuild the project and now it is working.
Thx for the fast support

OwensDataCenter, 2022-06-30:
@martin-ger How do I exactly do that?

axmed2004, 2022-09-20:
Please help me, how to build project?

martin-ger, 2022-09-27:
https://github.com/martin-ger/esp32_nat_router#building-the-binaries-method-1---espidf

SoothingVoid, 2023-02-12:
I am facing similar problem as related to uplink Wifi password, that is 63 char long, it just is not saved, will try suggested HTTPD_MAX_REQ_HDR_LEN fix.

Another thing I noticed that there might be some issues with special characters, "+" gets replaced with space(" ").

SoothingVoid, 2023-02-12:
I am facing similar problem as related to uplink Wifi password, that is 63 char long, it just is not saved, will try suggested HTTPD_MAX_REQ_HDR_LEN fix.

Another thing I noticed that there might be some issues with special characters, "+" gets replaced with space(" ").
Tried with set_sta and from webpage, both seem to do so.

4ybaka, 2025-03-02:
I managed to workaround it with curl. Just run 2 times next command to configure SSID and AP:
`curl '<here is url from browser that fails with 431>'`