martin-ger/esp32_nat_router #28: No internet connection?

starkapandan, 2020-09-08:
Hi, first of all thanks for the project and the simplicity of flashing it through a cmakelists.txt that works with vscode esp-idf in windows.

Everything is up and running, i can see the repeater has connected to my router, and an access point of the repeater has been created.

When i use the access point however there is no internet at all, why is that? There is no log or something similar, just when i connect to the access point it shows "I (422356) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2" which seems all good. So all links are connected.

Sidenote: My router aka the station uses 192.168.1.1 etc and the esp32 is set to 192.168.4.1 in the accesspoint network maaaybe if that has something to do with it?

Thanks before hand if you have the time to answer this post 👍

starkapandan, 2020-09-08:
More info: 
* I have tried setting ip adress to match 192.168.1.1 on access point, did not work.
* Tried settings dns to 192.168.1.1, did not work.

tichaonax, 2021-01-14:
I have the same issue, it connects but does not forward traffic to the internet.

I (214507) wifi:station: b8:d7:af:10:7e:16 join, AID=1, bgn, 2010,0>, prof:1
I (214577) ESP32 NAT router: 1. station connected
I (215067) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
esp32> 
esp32> show
I (284177) cmd_router: ssid *****
I (284177) cmd_router: passwd ******
I (284177) cmd_router: ap_ssid *****
I (284177) cmd_router: ap_passwd *******
STA SSID: TP-Link_521B Password: *******
AP SSID: Hwandaza_Repeater Password: ******
Uplink AP connected
1 Stations connected
esp32> I (318057) wifi:new:<10,0>, old:<10,0>, ap:<10,2>, sta:<10,0>, prof:1
I (318057) wifi:station: aa:da:a6:00:da:55 join, AID=2, bgn, 20
I (318147) ESP32 NAT router: 2. station connected
I (318587) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.3

Has anyone figured out yet. Same behavior on the 8266 version. I prefer the esp32 as I can actually build with VSCode IDE with defaults so I do not have to go to the webserver.

tichaonax, 2021-01-14:
I think the default binaries are compiled with the NAT forwarding off. If you rebuild withe the following flags set it will forward. They were previously commented out in the "sdkconfig" file

# CONFIG_LWIP_IP_FORWARD  is not set
# CONFIG_LWIP_IPV4_NAPT  is not set

CONFIG_LWIP_IP_FORWARD=y
CONFIG_LWIP_IPV4_NAPT=y 

Once you fix that and recompile it will now carry the flags into the build. It works for me now

tichaonax, 2021-01-14:
Unfortunately cannot compile the esp8266 so he will have to recreate valid binaries

martin-ger, 2021-01-15:
I have no issues using the current binary.

And in the sdkconfig file all neccessary options are also set.

tichaonax, 2021-01-15:
> I have no issues using the current binary.
> 
> And in the sdkconfig file all neccessary options are also set.

Strange, I had to make changes to the config and it worked, The esp8266 I was able to verify it is good. It is an amazing piece of work you did there. Any time line to port the mesh functionality to ESP32 the speed is faster there.

I am not able to compile esp8266 in VSCode, I could have created binaries with defaults configurations.

Skywalkerf34, 2021-01-20:
Hi Martin-ger
I face the same problem using downloaded binary, flashing work but no internet trough ESP32
I didn't try to recreate the binary yet

Skywalkerf34, 2021-01-21:
Unfortunately, I can't recreate those by myself, can you kindly provide?

Skywalkerf34, 2021-01-24:
got the "right bin" at https://github.com/dabare/esp32_nat_router
now this work very well, can use VPN (running on the computer) and pass  :-)
Thanks a lot for this nice project

lole-elol, 2021-02-17:
Hey @Skywalkerf34 i tried it with your linked binaries and the orignal ones from @martin-ger. 
Sadly i only get the following:

[image]

Which results in me being able to connect to the AP but then not being able to get internet connection through the Station.
Since everybody seems to be able to get internet connection does anyone has a detailed explanation on how to get this running?

Skywalkerf34, 2021-02-18:
Hi,
I would suggest to check carefully about your credential and Capital letter inside as the firmware is running and/or be closer to your router as the embedded antenna on ESP32 are not so good