martin-ger/esp32_nat_router #31: Beginners Question: Not flashable - problem with the bin files?

djjayballz, 2020-12-15:
Hi i tried to flash two ESP32 with the ESPDownloadTool v3.8.5 with your settings.
`UnicodeDecodeError: 'gb2312' codec can't decode bytes in position 18-19: illegal multibyte sequence`
is the response. 

I used Putty for the serial output:
`
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
`

Maybe this has something to do with your last commit?
I also tried an older version of the DownloadTool.

Alternatifely, I dont get the esptool to install on linux nor on windows.

Any ideas?

martin-ger, 2020-12-15:
Did you download the binaries separately with the browser. Perhaps you corrupted them during download. Please try to download the zip-file and extract the binaries from there. This is the safer way...

sedanurP, 2021-01-12:
Hey. I have that same problem. Did you find any solution?

martin-ger, 2021-01-12:
The binaries are okay - please try the zip download and in case of trouble a lower baudrate for flashing.

sedanurP, 2021-01-12:
I did what you said but still:
When I try to flash it in flash download tool v2.3 it gives no surch file or directory error. (for boot files)
When I try to flash it in flash download tool v3.8.5, UnicodeDecodeError: 'gb2312' codec can't decode bytes in position 16-17: illegal multibyte sequence error.

Skywalkerf34, 2021-01-20:
Hi, it works for me with those binaries

martin-ger, 2021-04-05:
Probably a download error - try to download the complete repo as zip and extract the binaries from there.

HonkHansen, 2021-04-05:
Speculation. I for my part did not download the archives again but succeeded using the mentioned flag. I noticed this has been active for the ESP8266 variant.

I hope we can get others to validate one of the approaches to have a solution at hand.