martin-ger/esp32_nat_router #32: Automesh in esp32? just like esp8266.

Apurv505, 2020-12-22:
Will there be an automesh feature in near future? 
Esp32 is much more powerful and its automesh must work better than esp8226's.
Thank you for your work!!

martin-ger, 2020-12-22:
Currently, I am not working on that. But actually, it should not be so hard, as NAT-routing is now supported by the ESP-IDF.

Apurv505, 2020-12-22:
Being a newbie , it looked pretty complex to me, but I will try to do that. 
Thank you!

alexsandermsm, 2021-01-05:
if you can do autoesh.
let me know.
I need it.
please!

italocjs, 2021-11-22:
Has there been a development on this? ive spend all day trying esp-idf examples but none of them worked.  I dont care about speed, just need to send a couple of messages over websocket from any node in the mesh.