martin-ger/esp32_nat_router #33: WT32-ETH01 Support?

19leva89, 2021-01-09:
Good afternoon, can you make a NAT Router from the WT32-ETH01 board using your firmware?
[image]

martin-ger, 2021-01-09:
I'm sure, it can be done easily, but the current software has no Ethernet support. Interesting board - will try to get one...

19leva89, 2021-01-09:
OK, thanks, I will wait with impatience!

beta-tester, 2021-02-16:
i am also interested in that board and combining with your _nat_ project.
i still waiting for the shippment of that board.
in the meanwhile i found the following example, maybe that is a starting point...
https://github.com/espressif/esp-idf/tree/master/examples/ethernet/eth2ap

grewhit25, 2021-02-16:
I'm in, watching this thread with interest.

My WT32-ETH01 board is still a month or so off delivery; but if you are impatient like me and want to get a bit of a head start, then the ENC28J60 available on quick turnaround [example code](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/enc28j60) may give a bit of space to do some prototyping in the meantime if you have an ESP32 lying around.

The downside is (it seems) the SPI interface can only be clocked at about 6 Mhz for stable results.

19leva89, 2021-02-16:
Good afternoon, an interesting example eth2ap! I flashed my WT32-ETH01,
there is a LAN8720 chip, but it did not want to work.

beta-tester, 2021-02-16:
> Good afternoon, an interesting example eth2ap! I flashed my WT32-ETH01, there is a LAN8720 chip, but it did not want to work

it doesn't work?! that's not good to hear!
maybe there are differences in GPIO usage:
https://github.com/espressif/esp-idf/blob/master/examples/ethernet/README.md#common-pin-assignments

19leva89, 2021-02-17:
Bit setting menuconfig, wi-fi work, but does not want to receive IP. I will
continue to investigate.

19leva89, 2021-02-18:
I suffered for two days, but I could not get IP.

ср, 17 февр. 2021 г. в 10:19, Дмитрий С <19leva89@gmail.com>:

r32190, 2021-02-18:
It is as expected and listed:

Wi-Fi station doesn't receive any IP via DHCP?
All Layer 3 (TCP/IP functions) on the ESP32 are disabled, including the SoftAP DHCP server. This means that devices must be able to access another DHCP server (for example on a Wi-Fi router connected via ethernet) or should use statically assigned IP addresses.

beta-tester, 2021-02-18:
do you mean the WT32-ETH01 does not get an IP or the device that is connected to the AP?
as far as i understrood, the WT32-ETH01  itself only forward the traffic between ETH and AP (like a virtual cable). so no IP for the WT32-ETH01 at both ends. only the connected device should get an IP, as far the router at the ETH end is serving DHCP.

19leva89, 2021-02-18:
The device that is connected to the AP dont get IP.

beta-tester, 2021-02-22:
this is the schematics i got:
[WT32_ETH01_V2 schematic.pdf](https://github.com/martin-ger/esp32_nat_router/files/6025280/WT32_ETH01_V2.schematic.pdf)

19leva89, 2021-02-23:
Oh thank you, I'll keep myself!

ericfont, 2021-12-08:
FYI, at the bottom of https://esphome.io/components/ethernet.html#configuration-for-wireless-tag-wt32-eth01 you will see the configuration settings for WT32-ETH01:

```
Configuration for Wireless Tag WT32-ETH01 ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16
```

wizarq, 2021-12-09:
Hi,

i subscribed to this issue too, as I am having the same issues with the WT32-ETH01 from wireless-tag.com as well (ESP-IDF v4.3.1-dirty version); it will not receive any dynamic IP address on an ethernet network:

```
I (5431) eth_example: Ethernet Link Up
I (5441) eth_example: Ethernet HW Addr 7c:87:ce:44:d1:8f
D (5771) emac_esp32: receive len= 130
D (5961) emac_esp32: receive len= 78
D (5981) emac_esp32: receive len= 342
D (6301) emac_esp32: receive len= 130
D (6961) emac_esp32: receive len= 150
D (6981) emac_esp32: receive len= 90
D (7061) emac_esp32: receive len= 90
D (7261) emac_esp32: receive len= 150
D (8761) emac_esp32: receive len= 342
```

I also noticed that there is no DHCP request coming from the device..
DHCP Server tcpdump output (exclude 192.168.6.105, as this is my host pc):

```
root@Teltonika-RUT240:~# tcpdump -i br-lan ! host 192.168.6.105 -e -n -vv
tcpdump: listening on br-lan, link-type EN10MB (Ethernet), capture size 262144 bytes
12:13:49.420807 00:1e:42:39:86:bd > 33:33:00:00:00:16, ethertype IPv6 (0x86dd), length 110: (hlim 1, next-header Options (0) payload length: 56) fe80::21e:42ff:fe39:86bd > ff02::16: HBH (rtalert: 0x0000) (padn) [icmp6 sum ok] ICMP6, multicast listener report v2, 2 group record(s) [gaddr ff05::1:3 to_ex { }] [gaddr ff02::1:2 to_ex { }]
12:13:50.120789 00:1e:42:39:86:bd > 33:33:00:00:00:16, ethertype IPv6 (0x86dd), length 110: (hlim 1, next-header Options (0) payload length: 56) fe80::21e:42ff:fe39:86bd > ff02::16: HBH (rtalert: 0x0000) (padn) [icmp6 sum ok] ICMP6, multicast listener report v2, 2 group record(s) [gaddr ff05::1:3 to_ex { }] [gaddr ff02::1:2 to_ex { }]
12:13:50.414952 00:1e:42:39:86:bd > 33:33:00:00:00:01, ethertype IPv6 (0x86dd), length 142: (hlim 255, next-header ICMPv6 (58) payload length: 88) fe80::21e:42ff:fe39:86bd > ff02::1: [icmp6 sum ok] ICMP6, router advertisement, length 88
        hop limit 64, Flags [managed, other stateful], pref medium, router lifetime 0s, reachable time 0s, retrans time 0s
          source link-address option (1), length 8 (1): 00:1e:42:39:86:bd
            0x0000:  001e 4239 86bd
          mtu option (5), length 8 (1):  1500
            0x0000:  0000 0000 05dc
          rdnss option (25), length 24 (3):  lifetime 18000s, addr: fe80::21e:42ff:fe39:86bd
            0x0000:  0000 0000 4650 fe80 0000 0000 0000 021e
            0x0010:  42ff fe39 86bd
          dnssl option (31), length 24 (3):  lifetime 18000s, domain(s): lan.
            0x0000:  0000 0000 4650 036c 616e 0000 0000 0000
            0x0010:  0000 0000 0000
          advertisement interval option (7), length 8 (1):  1800000ms
            0x0000:  0000 001b 7740
```

ygoe, 2022-02-20:
@ericfont Where would I put this in a regular esp-idf project? I have an "sdkconfig" file but it looks different from the code above.

beta-tester, 2022-02-21:
> @ericfont Where would I put this in a regular esp-idf project? I have an "sdkconfig" file but it looks different from the code above.

i hope i don't tell just absolute bullshit, but i think:
- this format of config parameters are not made for `sdkconfig`.
  i guess you have to run `make menuconfig` or `idf.py menuconfig`, find and navigate to those options and set them there.
  for possible names of options, maybe take a look to [eth2ap](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/eth2ap), [enc28j60#pin-assignment](https://github.com/espressif/esp-idf/tree/master/examples/ethernet/enc28j60#pin-assignment) or [eth2wifi#rmii-phy-wiring](https://github.com/espressif/esp-iot-solution/tree/release/v1.0/examples/eth2wifi#rmii-phy-wiring)
- but even then, i don't think that there is the ETH part implemented in this project code at all, yet.

i would be happy when i am wrong and the ETH part is implemented and working already

jokerjmv, 2022-08-30:
> FYI, at the bottom of https://esphome.io/components/ethernet.html#configuration-for-wireless-tag-wt32-eth01 you will see the configuration settings for WT32-ETH01:
>
> ```
> Configuration for Wireless [...]

Hello, I spend several hours trying to connect the board to the internet using the ETH. Be aware that the GPIO0 is used for the ETH interface and to select the Boot mode, so after programming the module, leave this pin disconnected.

tifou92, 2022-09-14:
> Hello, I spend several hours trying to connect the board to the internet using the ETH. Be aware that the GPIO0 is used for the ETH interface and to select the Boot mode, so after programming the [...]

Hello,
So, we the GPIO0 pin disconnected, you have the ap2eth working?

jokerjmv, 2022-09-14:
Yes, once I download the code to the WT32-ETH01 I disconnect the GPIO0 from the programmer board and it works. Don't forget to restart the module after programming.

tifou92, 2022-09-15:
What do you put in  the "Support ESP32 internal EMAC controller" submenu config?
It doesn't work for me and I don't understand why now...

jokerjmv, 2022-09-15:
I'm using Arduino framework and the first test was with the examples in the Arduino IDE, and later I developed my custom application using Platformio and Arduino framework as well.I never used the ESP32 configuration tool.

tifou92, 2022-09-15:
> Yes, once I download the code to the WT32-ETH01 I disconnect the GPIO0 from the programmer board and it works. Don't forget to restart the module after programming.

Something strange with mine...
After I connect my phone to the AP, when I scan my network, I just find 4 on 26 devices... And I can't access to Internet...
I can't uderstand why...

beta-tester, 2022-10-04:
only for information:
there is a similar device available to the WT32-ETH01.
the [PoESP32](https://docs.m5stack.com/en/unit/poesp32). with the different, it can be powered over ethernet.

rrwakc, 2023-02-13:
Olimex also has 2 ESP32 POE boards and can be ordered at mouser or similar distributors

https://eu.mouser.com/ProductDetail/Olimex-Ltd/ESP32-POE-ISO?qs=sGAEpiMZZMuqBwn8WqcFUj2aNd7i9W7uc087HzBKguU1UBkflb3j3w%3D%3D

https://eu.mouser.com/ProductDetail/Olimex-Ltd/ESP32-POE?qs=sGAEpiMZZMuqBwn8WqcFUj2aNd7i9W7u96Xy%252Be9XGB1dNrI6ppx4Mg%3D%3D

beta-tester, 2023-02-13:
another candidat with eth:
- [LilyGo T-Internet-COM](https://github.com/Xinyuan-LilyGO/T-Internet-COM)
- [LilyGO T-ETH-POE](https://github.com/Xinyuan-LilyGO/LilyGO-T-ETH-POE)

olloESP, 2023-05-11:
Hi,

I'd be interested in using the WT32-ETH01 as a NAT router - any progress on the implementation of the ETH support?

I do have the need to run the WiFi in Client mode on a different subnet than the eth. Eth is connecting as a client to the Internet router, getting IP through DHCP. All data from Eth shall be routed to the WiFi and vice versa.

I hope the project can be extended to support the ETH port.

jano1276, 2023-11-19:
Hello,
next one who is interested to ethernet support :)

Regards,
Jan

ortegafernando, 2024-02-21:
Hi, any option to also work with a W5500 module? Thanks

martin-ger, 2024-02-22:
Sorry, not so far.

--18dcf9df1843ec727ae69b14b0
Content-Type: text/html; charset="UTF-8"
Content-Transfer-Encoding: quoted-printable

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 4.01 Transitional//EN" "http://www.=
w3.org/TR/html4/loose.dtd">
<html>
<body>
<div dir=3D"auto">
<div dir=3D"auto">Sorry, not so far.</div><div dir=3D'auto'><br></div>
<div id=3D"aqm-original" style=3D"color: black;">

<div class=3D"aqm-original-body">
<div style=3D"color: black;">
<p style=3D"color: black; font-size: 10pt; font-family: sans-serif; margin:=
 8pt 0;">Am 21. Februar 2024 22:05:05 schrieb ortegafernando &lt;notificati=
***@***.***&gt;:</p>
<blockquote type=3D"cite" class=3D"gmail_quote" style=3D"margin: 0 0 0 0.75=
ex; border-left: 1px solid #808080; padding-left: 0.75ex;">
<p></p>
<p dir=3D"auto">Hi, any option to also work with a W5500 module? Thanks</p>

<p style=3D"font-size:small;-webkit-text-size-adjust:none;color:#666;">=E2=
=80=94<br>
n-ger/esp32_nat_router/issues/33#issuecomment-1957811439">view it on GitHub=
C36AGKYKDOLF27F2QTYUZHOXAVCNFSM4V3TFBXKU5DIOJSWCZC7NNSXTN2JONZXKZKDN5WW2ZLO=
u commented.<img src=3D"https://github.com/notifications/beacon/AF2UTC7ZKGA=
XWNLCREWFHLTYUZHOXA5CNFSM4V3TFBXKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJK=
TDN5WW2ZLOORPWSZGOOSY5J3Y.gif" height=3D"1" width=3D"1" alt=3D""><span styl=
e=3D"color: transparent; font-size: 0; display: none; visibility: hidden; o=
verflow: hidden; opacity: 0; width: 0; height: 0; max-width: 0; max-height:=
/33/1957811439</span><span>@</span><span>github</span><span>.</span><span>c=
om&gt;</span></span></p>

</blockquote>
</div>
</div>

</div><div dir=3D"auto"><br></div>
</div></body>
</html>

--18dcf9df1843ec727ae69b14b0--

ortegafernando, 2024-02-23:
> […](#)
> --18dcf9df1843ec727ae69b14b0--

Hi, I have just read here that with Arduino Core 3.0.0. W5500 is natively supported. Does it change anything for getting a nat router with W5500 module? 

https://github.com/arendst/Tasmota/discussions/18207#discussioncomment-8565056

FulvioSpelta, 2024-08-07:
> Hi,
>
> I'd be interested in using the WT32-ETH01 as a NAT router - any progress on the implementation of the ETH support?
>
> I do have the need to run the WiFi in Client mode on a different subnet than [...]

Any news about wifi-eth support ?
I'm available for testing if help is needed.
Thanks

rburkovsky, 2025-01-10:
> Hi, any option to also work with a W5500 module? Thanks

Also interested in this, would be awesome to have that supported.

T0rchw00d, 2025-01-21:
I'm also interested in getting the WT32-ETH01 up and running with Ethernet support as a bridge. I have a very small gadget with only one Ethernet port that I want to connect wirelessly to my router. Any other solution seems too big to fit in the case.

Volker

rburkovsky, 2025-01-21:
> I'm also interested in getting the WT32-ETH01 up and running with Ethernet support as a bridge. I have a very small gadget with only one Ethernet port that I want to connect wirelessly to my router. Any other solution seems too big to fit in the case.
> 
> Volker

@T0rchw00d I did exactly that with Orange Pi Zero 3 + openwrt to bring internet back to my old TV.
But I originally planned this project on ESP32, would be awesome.

T0rchw00d, 2025-01-23:
Yes, I had already tried alternatives: a RockPi-S and an OrangePi zero. Ultimately, neither of there wifi chips were supported by openWRT.
They are also too expensive, use too much power and are also quite large. An ESP32-ETH1 would be ideal and would be completely sufficient in terms of data throughput.
At last, I don't want to spend any more money on it.

rburkovsky, 2025-01-23:
> Yes, I had already tried alternatives: a RockPi-S and an OrangePi zero. Ultimately, neither of there wifi chips were supported by openWRT. They are also too expensive, use too much power and are [...]

Can't speak about Zero, but I was able to install Openwrt on OPi Zero 3 and set it up. There's no official support, but there's a version of Openwrt on github that has wifi working for this chip. It was really frustrating though and this board is kinda overkill for my purpose. ESP32 with Ethernet and WiFi would be ideal.