martin-ger/esp32_nat_router #34: Google DNS inaccessible in China

tibochina, 2021-01-16:
Thank you very much for this awesome application !
I can successfully flash and setup the ESP32 but for some reasons in can’t pen any websites.
I suspect this is because of the DNS being set to google 8.8.8.8 DNS, because google is blocked in China.
The problem is because of my total lack of experience I’m totally unable to change the bin files.

Is there a possibility to provide the bin files with another DNS please? Or even the possibility to set the DNS in the setting webpage without having to change the program and recompile everything? 
thank you in advance for your consideration!

nissimzur, 2021-01-16:
This is a wireless NAT. Connected to existing home router. No need for any DNS on this app. If your home router has working DNS all will work OK.

tibochina, 2021-01-16:
Thank you for your answer.
Then if the DNS is not a problem, any suggestions about where the problem could be from?
Everything seem to be just fine.
I can connect to the newly created extended wifi and the computer shows that I am connected to the internet (I could also successfully ping a website) but when try to open a website with a web browser it tries to download forever but the webpage does not load.
The result is the same for the computer and the tablet.

Any suggestions?

martin-ger, 2021-01-16:
This is an issue, as the ESP always hands out the MY_DNS_IP_ADDR  as DNS server for clients (currently 8.8.8.8), it currently does not change this, as soon as it learns a local DNS from its uplink DHCP server. As quick fix you might try to set a know working static DNS server in your client, but of course this is not desirable for dynamic clients.

I will try to update this in the next time...

martin-ger, 2021-01-16:
Easier than I thougt...

Just pushed a new version.

tibochina, 2021-01-16:
Awesome! You totally fixed the issue.
Thank you very much!!

As a side note, I noticed that I cannot connect to my VPN when I am connected to the esp32 NAT router. I have seen that someone else had opened another issue regarding this VPN problem half a year ago but there have been no comments about it.
Does it mean that the NAT isn’t compatible with VPN connections?

martin-ger, 2021-01-16:
Depends on VPN technology. What kind of VPN did you try?

tibochina, 2021-01-16:
I use Algo VPN from [https://github.com/trailofbits/algo](url)

martin-ger, 2021-01-16:
It seems to use IPSec or Wireguard. At least IPSec is known not to work over NAT without additional means. Wireguard also needs some config as far as I can see, but I am not really into it.

SSL VPN should work without problem and further config...

tibochina, 2021-01-17:
Oh I see thank you very much.
Your application is super cool, thank you!

closed 2021-01-17 by tibochina