martin-ger/esp32_nat_router #36: i this project able to "nat" between 802.11-B and 802.11-LR (Long Range)?

beta-tester, 2021-02-16:
hi,

i am wondering, if it is possible to do not only do "nat" between same wifi standards. would it be possible to to "nat" between wifi standard 80.2.11-B and [802.11-LR](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#long-range-lr) with one and the same ESP32?

so my PC can communicate with its wifi to the ESP32 via standard 802.11-B
and the ESP32 would move the network packets from the PC to the ESP32 sensors i have via the ESP32 propriatary wifi standard [802.11-LR](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#long-range-lr), that can reach a theoretical line of sight distance of about 1km at low speed.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#lr-compatibility

martin-ger, 2021-02-16:
I am pretty sure, that the ESP's radio can only operate in one mode at a time - it is the same with b/g/n. In StationAP mode it must be able to receive frames from the STAs at the same time as from the AP. This cannot be done with two different PHY settings at the same time,

beta-tester, 2021-02-17:
hmm... then maybe with the WT32-ETH01 board. it has an ethernet connector i can plug to the PC/switch and then the ESP32 (WT32-S1) is doing the 802.11-LR wifi standard to receive the sensors...

closed 2021-02-22 by beta-tester