martin-ger/esp32_nat_router #39: NAT and UDP packets

slater0013, 2021-03-04:
Hello,

Trying to use esp32_nat_router with a big wifi antenna connected to a WROOM_32U to extend my Drone range.

Installation is ok, configuration is ok.

But my drone is sending the video stream as an UDP feed. And I suppose the NAT router does not know to which client send it !

Is there any way to specify a "DMZ" NATed client that would receive every incoming packets, including UDP packets that are not in the NAT table.

Hope It's clear ?!

Thanks,
GS.

martin-ger, 2021-03-05:
No, there is no default client. You can define a portmap rule for any required UDP port.

slater0013, 2021-03-05:
Ok ! Any Link to a documentation in order to achieve such rules ?
Thanks !

slater0013, 2021-03-13:
Sorry for asking again, any hints how to add such port redirections ?

martin-ger, 2021-03-13:
You have to modify the  code: add the call(s) to ip_portmap_add() after esp32_nat_router.c#L367

skorokithakis, 2024-05-27:
A DMZ mode would be great, as I'm also running a single client behind this. I don't know how much work a fully-transparent "repeater" mode would be (ie it just extends the current LAN by transporting packets back and forth, without NAT), but that would be ideal.