martin-ger/esp32_nat_router #41: Header fields are too long for server to interpret

ChangboBro, 2021-03-07:
After setting my SSID and password in STA settings, click "Connect" and this strings come out on my screen.
I guess it's because my SSID contains Chinese characters, do I need to change my WiFi's name? I will try later, if it works, I will comments below.
Also the board I used for this firmware is ESP32 CAM, I don't know whether it is compatible for this.

ChangboBro, 2021-03-07:
I have tried to change my existing WiFi's SSID into full English, but nothing changed. :(

martin-ger, 2021-03-07:
ESP32 Cam should be no issue.

Did ou try it like this: use HTTP-style hex encoding like "Mine%2bYours". This will result in a string "Mine+Yours". With this hex encoding you can enter any byte value you like, except for 0 (for C-internal reasons)?

ChangboBro, 2021-03-09:
Oh, I found the issues! The first time I opened the web interface using my smartphone's native browser, and the problem occurred, then I tried Chrome browser, and it works totally fine!
Maybe it's because the native browser of my smartphone redirects or refreshes the web page immediately after I click the 'set' button, it doesn't wait for the ESP32 to process the information?

Also, I changed my existing WiFi's SSID back into Chinese again, the STA settings of the web interface can set WiFi's SSID into Chinese, but it seems the ESP32 can't connect to it. So maybe I should use English SSID now.

ChangboBro, 2021-03-09:
> HTTP-style hex encoding

Sorry, I didn't learn web development before, so I didn't understand what "HTTP-style hex encoding" means. but I find a website that introduces URL Encoding here：https://www.w3schools.com/tags/ref_urlencode.ASP
so I used the URL encoder on that web page to encode my WiFi's Chinese SSID. Fortunately, somehow it works! Thank you for your reply and your project! It helps me a lot.

ChangboBro, 2021-03-09:
> ESP32 Cam should be no issue.
> 
> Did ou try it like this: use HTTP-style hex encoding like "Mine%2bYours". This will result in a string "Mine+Yours". With this hex encoding you can enter any byte value you like, except for 0 (for C-internal reasons)?

Sorry, I didn't learn web development before, so I didn't understand what "HTTP-style hex encoding" means. but I find a website that introduces URL Encoding here：https://www.w3schools.com/tags/ref_urlencode.ASP
so I used the URL encoder on that web page to encode my WiFi's Chinese SSID. Fortunately, somehow it works! Thank you for your reply and your project! It helps me a lot.

closed 2021-03-09 by ChangboBro