martin-ger/esp32_nat_router #44: ESP32 don't connect to uplink WiFi on Channels 12/13 ?  /  Setting ESP32 WiFi-Bandwidth 20/40MHz ?

CRprog, 2021-04-11:
[1]
When I use my home AP on channels 1-11, there is an uplink connection quickly, everything works fine.
If I set my home AP to channel 12 or 13 (Fritzbox, 20MHz bandwidth), no uplink connection succeeds, never.
Is there a limitation or is it default restricted to US channels (1-11) ? 
(esp_wifi_set_country(WIFI_COUNTRY_xxx)
` WIFI_COUNTRY_CN = 0,   /**< country China, channel range [1, 14] */`
`WIFI_COUNTRY_JP,       /**< country Japan, channel range [1, 14] */`
`WIFI_COUNTRY_US,      /**< country USA, channel range [1, 11] */`
`WIFI_COUNTRY_EU,      /**< country Europe, channel range [1, 13] */`
`WIFI_COUNTRY_MAX`
`WIFI_COUNTRY_POLICY_AUTO,   /**< Country policy is auto, use the country info of AP`

[2]
Is it possible to configure the channel bandwidth down to 20MHz for the ESP32-AP ? (Defaults to 40MHz)
That gives me more room for a optimized channel distribution (1/5/9/13).

martin-ger, 2021-04-11:
[1] There is no explicit setting for the WIFI_COUNTRY, so according to the manual the default should apply, which should be fine for your setting:

> 1. The default country is {.cc=”CN”, .schan=1, .nchan=13, policy=WIFI_COUNTRY_POLICY_AUTO} 

[2] In STATION_AP mode, which is used here, the WiFi settings for the ESP's AP are always the same as for the STA and thus determined by the uplink AP. You cannot change this as there is only one transceiver.

CRprog, 2021-04-11:
[1] ** Solved **
So I did a little more research and found a strange cause:
Until now I had used a hidden SSID for my Home-AP (I know, not recommended ...) 
This did not cause any problems on channels 1-11, but obviously on channels 12 and 13.
I've change this now to visible SSID and the ESP32 connected immediately on channels 12 or 13.

[2]
Here is the typed output of the "Wifi Analyzer" Access-points from my cell phone :
Home-AP (00:1f:3f:xx:xx:xx)
-40dBm Ch 13 2472MHz  ~1,0m
2462-2482 (**20MHz**) AVM GMBH

ESP32-AP (30:ae:a4:xx:xx:xx)
-51dBm Ch 13(11) 2472MHz  ~3,4m
2442-2482 (**40MHz**) ESPRESSIF IN

The ESP32-AP is apparently for the "ESP32 NAT Router Config" - page an 192.168.4.1
Then I will deactivate over CLI the (new nice) webinterface, which really doesn't need that high 40MHz bandwidth,
and free up a lot of channels. 
Maybe a practical new future-feature: Deactivate webinterface automatically e.g. 15min after switiching on,
perhaps additionally a check mark in the web interface for that.

CRprog, 2021-04-23:
[2]
I've tried it now: As expected, deactivating the webinterface didn't change the ESP32 40MHz bandwidth to 20MHz. (Home-AP)
Not really important, but ..... command in the CLI to set the bandwidth would be great, or of course to make it automatically as wide as the home AP.

closed 2021-06-02 by CRprog