martin-ger/esp32_nat_router #46: no internet connectivity for a second router

afennani, 2021-04-13:
My ultimate goal is to have 2 esp32 acting as range extender for my home wifi.

**HOME ROUTER** <===(802.11 bgn)===> **Gateway (esp32)**<==(LR mode)==>**SPOT-1(esp32)**<=== esp32 iot devices
let's ignore the LR mode for now.
- 1 esp32 as Gateway: connect to my router and exposes AP (Gateway)
==> when i connect to the Gateway: i have internet and everything is working just fine
- if i setup a second esp32 (let s call it SPOT-1): i configure it to connect to Gateway as station And Exposes an AP called SPOT-1, i can see it gets the right configuration. 
- my laptop can connect to SPOT-1 but i can't get internet there. even though i m getting the dns from my home router
- i tried to change the ip addresses (Gateway on 192.168.4.1, and SPOT-1 on 192.168.5.1) with no luck
- am i missing something?

martin-ger, 2021-04-13:
Where did you change the IP address? Here: https://github.com/martin-ger/esp32_nat_router/blob/master/main/esp32_nat_router.c#L505 ?
Think, this should do the job...

martin-ger, 2021-04-13:
Added a new command on the serial console: "set_ap_ip <addr>".
With this you can modify the IP of the ESP's AP.

afennani, 2021-04-13:
thanks for the change. it will help for sure.
but my problem is not the ip itself. i had 2 builds before (one with 192.168.4.1 and the other with 192.168.5.1).
all the ip addresses are fine it s just when i connect the Gateway AP i have internet. but when i connect to SPOT-1 (which is connected to Gateway)