martin-ger/esp32_nat_router #48: which idf version?

xenpac, 2021-04-19:
hi,

i used idf 4.4, got:
WARNING: Missing submodule components/micro-ecc/micro-ecc...
Attempting 'git submodule update --init components/micro-ecc/micro-ecc' in esp-idf root directory...
error: pathspec 'components/micro-ecc/micro-ecc' did not match any file(s) known to git.
/home/tom/esp/esp-idf/make/project.mk:703: recipe for target '/home/tom/esp/esp-idf/components/micro-ecc/micro-ecc/.git' failed

which version did u use to compile?

xenpac, 2021-04-20:
ok, now im getting:
esp32_nat_router/components/cmd_router/cmd_router.c:27:2: error: #error "IP_NAPT must be defined"
 #error "IP_NAPT must be defined"
  ^~~~~

martin-ger, 2021-04-20:
Did you configure this:
```
in the project directory run make menuconfig (or idf.py menuconfig for cmake).

    *Component config -> LWIP > [x] Enable copy between Layer2 and Layer3 packets.
    *Component config -> LWIP > [x] Enable IP forwarding.
    *Component config -> LWIP > [x] Enable NAT (new/experimental).
```

xenpac, 2021-04-20:
ok thanks, i just used the binarys for now.

xenpac, 2021-04-20:
works;)
but the esp_nat_router appears as "non-encrypted" although i set a passwd in the webconfig.
do i have to select an encryption scheme somewhere?

martin-ger, 2021-04-20:
Should be encrypted after restart - perhaps something cached in the client?

xenpac, 2021-04-20:
what encryption scheme are you using?

martin-ger, 2021-04-20:
WPA-PSK

xenpac, 2021-04-20:
ok, a passwd length of 4 was too short ;_)
now using a length of 10 and it works.

xenpac, 2021-04-20:
is this project actually the same as the esp-wifi-repeater , just for the esp32?

martin-ger, 2021-04-20:
The idea is the same, but the implementation is completely different and it 
has LEDs features.

xenpac, 2021-04-20:
nebenbei....eine seehr schöne software...Kompliment!