martin-ger/esp32_nat_router #53: Feature request

megabyte2010, 2021-05-26:
I like your repeater code. It works very well. But it would be nice if we could scan nearby access points, and show some information about them, like ssid, dbm etc..

martin-ger, 2021-05-26:
Should be not so difficult to combine it with something like:
https://github.com/VALERE91/ESP32_WifiScan

If I (or somebody else) find some time.

 

TLIFANCNFSM45RZ5DCA> .
<https://github.com/notifications/beacon/AF2UTC4AQGQPKAKTMRZSHDTTPTLIFA5CNFS
M45RZ5DCKYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4NOHC4AQ.gif>

dchristl, 2022-03-26:
> I like your repeater code. It works very well. But it would be nice if we could scan nearby access points, and show some information about them, like ssid, dbm etc..

Hi @megabyte2010 , I missed the same functionality and used this project as a starting point for learning microcontroller programming. Your requested feature is already implemented:

[image]

Feel free to have a look at [my project](https://github.com/dchristl/esp32_nat_router_extended).