martin-ger/esp32_nat_router #55: STA Static IP not working

01yakiek, 2021-06-07:
The esp32 setup connects to the internet without setting a static ip.
But as soon as I set it, connection to the internet stops.

I checked my dhcp clients, and the device "expressif" ip is not set to "192.168.0.9".
On my computer wifi, setting static ip works, but just not on the esp32 for some reason.

Did I do something wrong?

Here is the log:
<details>
 <summary>Click to expand log</summary>

```
00:22:27.240 -> esp32> set_sta_static  192.168.0.9 255.255.255.0 192.168.0.1
00:23:12.993 -> [0;32mI (251407) cmd_router: STA Static IP settings 192.168.0.9/255.255.255.0/192.168.0.1 stored.[0m
00:23:12.993 -> esp32> [0;32mI (265727) HTTPServer: Found header => Host: 192.168.4.1[0m
00:23:29.873 -> [0;32mI (268307) HTTPServer: Found header => Host: 192.168.4.1[0m
00:23:33.075 -> restart
00:23:33.406 -> [0;32mI (271857) cmd_system: Restarting[0m
00:23:33.406 -> I (271857) wifi:state: run -> init (0)
00:23:33.452 -> I (271857) wifi:pm stop, total sleep time: 108057005 us / 268982778 us
00:23:33.452 -> 
00:23:33.452 -> I (271857) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<1,0>, prof:1
00:23:33.452 -> I (271867) wifi:station: [redacted] leave, AID = 1, bss_flags is 658531, bss:0x3ffd7da4
00:23:33.452 -> I (271877) wifi:new:<1,0>, old:<1,0>, ap:<1,1>, sta:<1,0>, prof:1
00:23:33.452 -> W (271877) wifi:W (271887) wifi:hmac tx: ifx0 stop, discard
00:23:33.452 -> <ba-del>idx
00:23:33.452 -> [0;32mI (271887) ESP32 NAT router: disconnected - retry to connect to the AP[0m
00:23:33.452 -> [0;32mI (271897) ESP32 NAT router: station disconnected - 0 remain[0m
00:23:33.500 -> [0;33mW (271907) httpd_txrx: httpd_sock_err: error in recv : 113[0m
00:23:33.546 -> I (271997) wifi:flush txq
00:23:33.546 -> I (271997) wifi:stop sw txq
00:23:33.546 -> I (271997) wifi:lmac stop hw txq
00:23:33.546 -> ets Jun  8 2016 00:22:57
00:23:33.594 -> 
00:23:33.594 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
00:23:33.594 -> configsip: 0, SPIWP:0xee
00:23:33.594 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:23:33.594 -> mode:DIO, clock div:2
00:23:33.594 -> load:0x3fff0030,len:5356
00:23:33.594 -> ho 0 tail 12 room 4
00:23:33.594 -> load:0x40078000,len:13732
00:23:33.594 -> load:0x40080400,len:3412
00:23:33.594 -> entry 0x40080634
00:23:33.921 -> [0;32mI (354) cpu_start: Pro cpu up.[0m
00:23:33.921 -> [0;32mI (355) cpu_start: Starting app cpu, entry point is 0x40081538[0m
00:23:33.921 -> [0;32mI (341) cpu_start: App cpu up.[0m
00:23:33.921 -> [0;32mI (369) cpu_start: Pro cpu start user code[0m
00:23:33.967 -> [0;32mI (369) cpu_start: cpu freq: 160000000[0m
00:23:33.967 -> [0;32mI (369) cpu_start: Application information:[0m
00:23:33.967 -> [0;32mI (373) cpu_start: Project name:     esp32_nat_router[0m
00:23:33.967 -> [0;32mI (379) cpu_start: App version:      5d490e2-dirty[0m
00:23:33.967 -> [0;32mI (385) cpu_start: Compile time:     Apr 13 2021 18:01:34[0m
00:23:33.967 -> [0;32mI (391) cpu_start: ELF file SHA256:  4d6fb183f7eaecc5...[0m
00:23:33.967 -> [0;32mI (397) cpu_start: ESP-IDF:          v4.4-dev-744-g1cb31e509[0m
00:23:33.967 -> [0;32mI (403) heap_init: Initializing. RAM available for dynamic allocation:[0m
00:23:33.967 -> [0;32mI (410) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
00:23:34.013 -> [0;32mI (416) heap_init: At 3FFB8118 len 00027EE8 (159 KiB): DRAM[0m
00:23:34.013 -> [0;32mI (423) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
00:23:34.013 -> [0;32mI (429) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
00:23:34.013 -> [0;32mI (435) heap_init: At 40097A00 len 00008600 (33 KiB): IRAM[0m
00:23:34.013 -> [0;32mI (443) spi_flash: detected chip: generic[0m
00:23:34.013 -> [0;32mI (446) spi_flash: flash io: dio[0m
00:23:34.013 -> [0;32mI (451) cpu_start: Starting scheduler on PRO CPU.[0m
00:23:34.013 -> [0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
00:23:34.060 -> [0;32mI (538) ESP32 NAT router: Command history enabled[0m
00:23:34.060 -> [0;32mI (538) cmd_router: ssid [redacted][0m
00:23:34.060 -> [0;32mI (538) cmd_router: passwd [redacted][0m
00:23:34.060 -> [0;32mI (538) cmd_router: static_ip 192.168.0.9[0m
00:23:34.106 -> [0;32mI (548) cmd_router: subnet_mask 255.255.255.0[0m
[... 58 lines cut ...]
00:23:36.651 -> On Windows, try using Putty instead.
00:23:36.651 -> esp32> W (5328) wifi:<ba-del>idx
00:23:42.566 -> I (9058) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
00:23:42.612 -> I (9058) wifi:station: [redacted] join, AID=1, bgn, 20
00:23:42.659 -> [0;32mI (9128) ESP32 NAT router: 1. station connected[0m
00:23:42.799 -> [0;32mI (9288) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2[0m
00:23:45.871 -> W (12318) wifi:<ba-add>idx:0 (ifx:0, [redacted]), tid:0, ssn:3, winSize:64
00:23:46.583 -> W (13068) wifi:<ba-add>idx:4 (ifx:1, [redacted]), tid:0, ssn:13, winSize:64
00:25:02.784 -> show
00:25:03.116 -> [0;32mI (89588) cmd_router: ssid [redacted][0m
00:25:03.116 -> [0;32mI (89588) cmd_router: passwd [redacted][0m
00:25:03.116 -> [0;32mI (89588) cmd_router: static_ip 192.168.0.9[0m
00:25:03.116 -> [0;32mI (89598) cmd_router: subnet_mask 255.255.255.0[0m
00:25:03.116 -> [0;32mI (89608) cmd_router: gateway_addr 192.168.0.1[0m
00:25:03.163 -> [0;32mI (89608) cmd_router: ap_ssid [redacted][0m
00:25:03.163 -> [0;32mI (89608) cmd_router: ap_passwd [redacted][0m
00:25:03.163 -> STA SSID: [redacted] Password: [redacted]
00:25:03.163 -> AP SSID: [redacted] Password: [redacted]
00:25:03.163 -> AP IP address: 192.168.4.1
00:25:03.163 -> Uplink AP connected
00:25:03.163 -> IP: 192.168.0.9
00:25:03.163 -> 1 Stations connected
00:25:03.163 -> esp32> 
```
</details>

Stasfe, 2021-11-19:
I have the same issue. Is there a solution for this?

maps2002, 2022-05-08:
Hi,
I also wanted to use@martin-ger great work but with a static IP on the STA.
I think the original code is not working because when a static ip is defined, the DNS is not properly configured.

I had success by changing a piece of code in file: esp32_nat_router.c, function: wifi_event_handler, for the switch/case of: SYSTEM_EVENT_STA_GOT_IP

This piece of original code:
```
if (esp_netif_get_dns_info(wifiSTA, ESP_NETIF_DNS_MAIN, &dns) == ESP_OK) {
    dhcps_dns_setserver((const ip_addr_t *)&dns.ip);
    ESP_LOGI(TAG, "DHCP: set dns to:" IPSTR, IP2STR(&dns.ip.u_addr.ip4));
}
```
I changed it to:
```
if (has_static_ip) {
    // Configure static DNS from DEFAULT DNS
    ip_addr_t dnsserver;
    dnsserver.u_addr.ip4.addr = ipaddr_addr(DEFAULT_DNS);
    dnsserver.type = IPADDR_TYPE_V4;
    dhcps_dns_setserver(&dnsserver);
    ESP_LOGI(TAG, "STATIC IP: set dns to:" IPSTR, IP2STR(&dnsserver.u_addr.ip4));
}
else {
    // Configure DNS from STA DHCP
    if (esp_netif_get_dns_info(wifiSTA, ESP_NETIF_DNS_MAIN, &dns) == ESP_OK) {
        dhcps_dns_setserver((const ip_addr_t *)&dns.ip);
        ESP_LOGI(TAG, "DHCP: set dns to:" IPSTR, IP2STR(&dns.ip.u_addr.ip4));
    }
}
```

In my case, I also updated the DEFAULT_DNS define, as I wanted to use my router ip address.
I am not using any of the portmap features, so I have not tested this change outside of my usage scope.

PS: The "new code" can probably be simplified to make the setserver and ESP_LOG instructions generic, and outside/after the if has_static_ip

Fretwire, 2022-08-20:
Here's a patch with the above code. Three sections:

First: Change default DNS to opennic dns server instead of Google
Second: Add (modified) above code to git source
Third: Update web page to hide STA password

[esp32_nr_patch.txt](https://github.com/martin-ger/esp32_nat_router/files/9386027/esp32_nr_patch.txt)

FrankNT, 2024-07-05:
Change those lines of code:
        esp_netif_dhcpc_stop(ESP_IF_WIFI_STA); // Don't run a DHCP client
        esp_netif_set_ip_info(ESP_IF_WIFI_STA, &ipInfo_sta);

To this:
        esp_netif_dhcpc_stop(wifiSTA); // Don't run a DHCP client
        esp_netif_set_ip_info(wifiSTA, &ipInfo_sta);

Work for me