martin-ger/esp32_nat_router #58: Static routing in mesh network

jigneshpatel47, 2021-08-03:
Hi,

I have created mesh on esp32c3 taken a reference from [wifi repeater](https://github.com/martin-ger/esp_wifi_repeater/issues). Thank you @martin-ger for this. 
We are using esp-idf revsion ESP-IDF v4.3-beta3-dirty. I have enable napt using ip_napt_enable(htonl(napt_netif_ip), 1); for packet forwarding.

I am able to ping and connect TCP port to root from a client that is connected to child node but I can not ping to child node from the client which is connected to root.
I have attached a mesh diagram for reference
[image]

Please help how to connect from root to child node.

Thank you,
Jignesh Patel

dherkes, 2021-08-03:
Thank you for this.

martin-ger, 2021-08-03:
192.168.3.2 is the child? You cannot connect in that direction through a NAT router. The only option is port forwarding but this has to be configured on all downstream routers and it can only be done for one device per port number.

This is a reason why I did the MQTT-VPN: https://github.com/martin-ger/MQTT_VPN

jigneshpatel47, 2021-08-03:
Yes, 192.168.3.2 are child node.

jigneshpatel47, 2021-08-04:
Hi, 

Is there any other way to connect the child from the root node? any idea will be appreciated.

Thank you,
Jignesh Patel

martin-ger, 2021-08-04:
As said above: to connnect to a node behind a NAT router you either need a port forward (difficult if you have several levels of NAT and many nodes):
`portmap  [add|del] [TCP|UDP] <ext_portno> <int_ip> <int_portno>`

or the connection is set up the other way round. That's why I started with MQTT-VPN: https://github.com/martin-ger/MQTT_VPN.

jigneshpatel47, 2021-08-04:
Thank you @martin-ger.

I will check into it.

Best Regards,
Jignesh Patel