martin-ger/esp32_nat_router #59: Is it works with ESP32-S2?

scriptedwedge, 2021-08-23:


scriptedwedge, 2021-08-23:
Hey, i tried to run it with ESP32 and it works and when i select ESP32-S2 on flash tool and flash that doesn't work. I tried with 3 S2's. but not sure why it won't work. Could you please bring me a light on this situation?

greyhatmiddleman, 2021-12-06:
So I too have an esp32-s2 from adafruit. In order to make this work you need to make sure the build target is for esp32s2. 

At a high level these were the steps that I took:

1. Install ESP-IDF, be sure to follow the steps accordingly
2. `cd` to esp32_nat_router folder
3. Delete the entire build folder
4. `get_idf` or run the export.sh as recommended in the ESP-IDF install
5. `idf.py set-target esp32s2`
6. In the `idf.py menuconfig`, make sure all the necessary settings are enabled as stated in https://github.com/martin-ger/esp32_nat_router#building-the-binaries
7. `idf.py build`
8. `idf.py -p PORT flash`

You might see an error message which should only be in regards on how it resets the controller. If the firmware reloaded successfully, you should be able to press the reset button and see the "ESP32_NAT_Router" SSID when you scan for wifi networks.

jonzobot, 2023-03-26:
These instructions worked for me. Unfortunately the throughput is really slow. I have a ESP32-S2-MINI-2U. I'm getting about 1.5-2 mbps. Not sure if it's my particular board or a config setting somewhere.

mikyk10, 2023-05-13:
It worked for me too. The only problem I stumbled upon was the `esp-idf` version. I had to use `esp-idf v4.x`.

lqSky7, 2024-10-27:
> It worked for me too. The only problem I stumbled upon was the `esp-idf` version. I had to use `esp-idf v4.x`.

Can you please share the binaries? When I do idf.py set-target esp32s2, it tries to clone something and never completes it. I've tried all available isps but can't get to work.