martin-ger/esp32_nat_router #62: how to cascade multiple repeaters

janjos0, 2021-09-29:
Let me start by thanking the creator of this project.
This is not an issue but a specific question.
I need to extend the WiFi signal of my internet router up to 280 meters.
The router assigns IP addresses in the range 192.168.0.xxx.
Because there is no power grid to power the repeater(s), they must be battery powered.
To solve my problem I thought of using several ESP32-WROOM-32 powered by 5V power-banks.
To try the set-up, I have in mind, I bought two ESP32 modules and flashed the code to them.
I set the STA settings (uplink WiFi network) of the first module, ~50 meters away from the router, to the SSID and passw. of my router. For the AP settings (the new network) I set ESP32-1 and a passw.
I set the STA settings of the second module, ~50 meters away from the first module, to ESP32-1 and passw. of the first. The AP settings (the new network) of the second I set ESP32-2 and same passw.
The STA Static IP Settings I leave them blank on both modules.
I see the first module (ESP32-1) connects to my router (the blue LED flashes) and the second connects to the first (the blue LED also flashes) and my laptop connects to the second (ESP32-2), but the laptop has no internet.
Questions:
Using this firmware is it possible to achieve my goal ?
If yes, what settings must I use to get internet in the laptop connected to the last repeater in the chain ?
Thanks for any replies.

janjos0, 2021-10-15:
@martin-ger
Thanks for the tip, it works.

@jbatx
Yes I did, like I wrote it works.
To get it working I changed the ap_ip of the second ESP32 NAT router.

My set-up is at follows:
Home router assigns IP addresses in the range 192.168.0.xxx
First ESP32 board connects directly to my home router and creates a WiFi network with ssid ESP32_1 and a passw.
I did not change it's ap_ip, it has the default 192.168.4.1.
Second ESP32 board connects to the first (ESP32_1) and creates a WiFi network with ssid ESP32_2 and a passw.
I changed its ap_ip address to 192.168.40.1, using PuTTY and the command "set_ap_ip 192.168.40.1" in the command line.
My laptop connects to the second (ESP32_2) and has internet. It works and I am writing this message using this set-up.

I still need to test the range, I need 280 meters, using just two ESP32-WROOM-32 boards, if needed I suppose I can add more.
I am no expert in networks, but I think the up and down speed does decrease by 50% by each added ESP32 NAT router.

Once again thanks to martin-ger.

muhammad-afiq, 2022-01-15:
@janjos0 
Hai, I want to have multiple repeaters as you do. How you change the IP of the second ESP32 using Putty? I don't know what is the configuration on putty to reach the second ESP32 board and enter the command.

janjos0, 2022-01-15:
@muhammad
1. Connect using MicroUSB cable the ESP32 board to your computer.
2. If a Windows machine, check Ports(COM&LPT) in "Device Manager" for your ESP32 COMx port, where x should be a number 1...99).
3. Start PuTTY, choose Serial as "Connection type", enter your "COMx" port number (e.g. COM8) in the "Serial line" box and 115200 in the "Speed" box.
4. Click Open.
5. In the new PuTTY connection window press the enter key.
6. If all went well, you should get esp32>
7. Enter help to get a list of available commands.
8. Enter command "set_ap_ip 192.168.xxx.1", replace the xxx with the number you want, e.g. 40 in my case (default is 4).
9. You should get a confirmation message "... AP IP address 192.168.xxx.1 stored.".
10. Close PuTTY (or not). If you leave it open you will see regular debug information being printed.

Connect your computer WiFi to the WiFi network created by the ESP32.
In your browser type in the address balk the IP address (e.g. 192.168.40.1) you entered in step 8 and press enter key.
You should get the graphical configuration page of the ESP32 NAT Router.

Everything is explained by Martin in the README here https://github.com/martin-ger/esp32_nat_router

muhammad-afiq, 2022-01-16:
@janjos0 
thanks for the steps, its working now.

jul-nc, 2023-07-27:
@janjos0 
Does it work too when I set at all ESP32's the same SSID and password?

janjos0, 2023-07-27:
@jul-nc
I did not try using same SSID on at least two of them and unfortunately I can not try it for the moment.
Mine had SSID's with only different last character like ESP32_1, ...2.
If you have two spare ESP32 boards you could try it and please let us know.