martin-ger/esp32_nat_router #66: How to make  a clean build

dherkes, 2021-11-21:
I have successfully built and flashed this software several times.  I do have a stupid question:  How do you make clean the files to build a fresh copy?  Thanks

martin-ger, 2021-11-22:
Does this answer the question?

From https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html :

- idf.py clean will “clean” the project by deleting build output files from the build directory, forcing a “full rebuild” the next time the project is built. Cleaning doesn’t delete CMake configuration output and some other files.

- idf.py fullclean will delete the entire “build” directory contents. This includes all CMake configuration output. The next time the project is built, CMake will configure it from scratch. Note that this option recursively deletes all files in the build directory, so use with care. Project configuration is not deleted.

dherkes, 2021-11-22:
On Sun, 21 Nov 2021 23:33:35 -0800
martin-ger ***@***.***> wrote:

> Does this answer the question?
>
> From
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html :
>
> - idf.py clean will “clean” the project by deleting build output [...]

It does.  Thank you so much.

-- 
Daniel J. Herkes Jr.
Sugar Grove, IL 60554