martin-ger/esp32_nat_router #70: Building on Windows with Platformio

mariusl, 2022-01-18:
Hi
I am not sure what your build environment is but I cannot figure out how to build this project using Visual Studio Code and PlatformIO. If there is a way to do it either with or without PlatformIO in VSC I would appreciate any help.

martin-ger, 2022-01-18:
It has been build using the plain ESP-IDF: https://github.com/martin-ger/esp32_nat_router#building-the-binaries

dchristl, 2022-03-26:
Hello mariusl,

you can check out my [modified version](https://github.com/dchristl/esp32_nat_router_extended) of this project. This is build with PlatformIO IDE extension of VSCode. The setup process is explained [here](https://github.com/dchristl/esp32_nat_router_extended/blob/master/docs/BUILD.md).

mariusl, 2022-03-26:
Wow that looks great. I opened the project and compiled straight away with no problems. The new interface looks great. Thanks for your hard work man, I am very exited about using your code now.
Just one question. Is there a way to upload from PlatformIO instead of using the upload tool?

dchristl, 2022-03-26:
Have you checked out my linked setup process documentation? In the last screenshot you can see the compile and upload button highlighted:

[image]

The 2nd one (arrow right) is for uploading to the device.

mariusl, 2022-03-26:
I did use that but the ESP just goes into a reset loop if I programmed with that. If I use the esptool with the same compiled binaries, it works ok. I don't know if I am missing a step somewhere?

mariusl, 2022-03-26:
This was the output from the loader. Note the last file location is not the same as what you specified in the document
`
Compressed 23136 bytes to 14287...
Writing at 0x00001000... (100 %)
Wrote 23136 bytes (14287 compressed) at 0x00001000 in 0.7 seconds (effective 258.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 116...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (116 compressed) at 0x00008000 in 0.1 seconds (effective 342.5 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 31...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (31 compressed) at 0x0000e000 in 0.1 seconds (effective 541.4 kbit/s)...
Hash of data verified.
Compressed 1225568 bytes to 675879...
`

mariusl, 2022-03-26:
Here is the complete output for the build. I don't see that the build.py script is running.
`
Processing esp32dev (platform: espressif32; board: esp32dev; framework: espidf)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (3.3.2) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf 3.40300.0 (4.3.0)
 - tool-cmake 3.16.4
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - tool-idf 1.0.1
 - tool-mconf 1.4060000.20190628 (406.0.0)
 - tool-ninja 1.9.0
 - toolchain-esp32ulp 1.22851.191205 (2.28.51)
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
WARNING: You are using pip version 21.2.4; however, version 22.0.4 is available.
You should consider upgrading via the 'C:\Users\mariu\.platformio\penv\Scripts\python.exe -m pip install --upgrade pip' command.
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Retrieving maximum program size .pio\build\esp32dev\firmware.elf
Checking size .pio\build\esp32dev\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  10.5% (used 34548 bytes from 327680 bytes)
Flash: [======    ]  58.4% (used 1225333 bytes from 2097152 bytes)
======================================================================= [SUCCESS] Took 41.04 seconds =======================================================================

`

dchristl, 2022-03-27:
I think your PlatformIO-Espressif 32 -platform module is too old (3.3.2). For support of your installed esp-idf (4.3.0) you need at least 3.4.0 (s [Release](https://github.com/platformio/platform-espressif32/releases))

[image]

I think after updating this will work. This is the only difference, beside of the pip warning, between your and mine output.

mariusl, 2022-03-28:
That did the trick. Thanks for your help, it's much appreciated.

dchristl, 2022-03-28:
This is called HTML minifying and is implemented inside the build.py-script with python.

mariusl, 2022-03-28:
Aah I am learning every day. Is it possible to put the un-minified scripts on repository. I want to add some functionality but it's a pain to edit the minified stuff. Although I have been doing manual reformatting with my CoffeeCup editor