martin-ger/esp32_nat_router #71: extremely slow/no network access

joedupa, 2022-01-18:
Hello, I flashed this project  onto ESP32 as well as ESP8266 project onto NodeMCU. In both cases I did not compile the projects but used pre-build binaries. Exactly the same problem persists for NodeMCU platform, but I will not discuss it here. I am using a ESP32 DEV KIT V1 from Amazon.  As I mentioned, I can successfully flash the board, access it's web site and configure to work with my home router. The board connects to the router, and I can successfully join the board's network. The first couple of external web sites load without any problem, but any subsequent ones choke to the point that browsing is no longer possible. When that happens, I can't even access my home router web interface, nor ping any devices on the home router's network. I have no idea how can I handle this issue, and hope that somebody here can guide me through the process of fixing the problem. Should the project be compiled for my specific board? or are the pre-compiled Github binaries sufficient? Again, exactly the same problem exists fo the ESP8266 platform. Please help. Thank you.

kunwarsethi, 2022-01-21:
Same, I have a Hiletgo ESP32, I can connect it to my router no problem but when a device connects to it, connection speeds are 150kb/s

HackerBeast, 2022-09-29:
Try setting SPI=80MHz and BAUD=921600

xiv3r, 2022-11-09:
set --flash_freq to 20M.