martin-ger/esp32_nat_router #72: Port Forward To STA Network

mtnbkr88, 2022-01-21:
Hi - I've been searching for a way to port forward to other IP/ports on the same network as the source IP/port. The reason... My router only supports a max of 20 port forwarding rules. I can set one rule to forward a range of ports to an internal network (192.168.2.X) IP device (currently a linux C.H.I.P. but hopefully to be an ESP32-CAM) and have that device forward those ports to other IP/ports on the same network (192.168.2.X) using linux iptables commands. I've not found any way to do this with the ESP32. The closest I've found is your esp32_nat_router.bin. This binary runs okay on my ESP32-CAM but it appears to only forward to the 192.168.4.X network. I have the static IP on the STA side set to 192.168.2.18. I used the command: portmap add TCP 7999 192.168.2.34 80 to forward port 7999 to an http server on the same network. Show shows: TCP 192.168.2.18:7999 -> 192.168.2.34:80 but the connection just times out. Does port forwarding only go from STA to AP? Is there any way I can use the ESP32 to forward ports in the same network? I appreciate the work you've done on this. Thank you.

martin-ger, 2022-01-23:
Port forwarding (and the esp32_nat_router) only works with different networks. NAT forwards from the STA side (outside, uplink) to the AP side (ESPs own network).

mtnbkr88, 2022-01-23:
Thank you for the reply. I found an option in opt.h called IP_FORWARD_ALLOW_TX_ON_RX_NETIF. I'm hoping setting this to 1 will allow the packet to forward out the incoming network like it says. Thank you.

closed 2022-01-23 by mtnbkr88