martin-ger/esp32_nat_router #74: Does the  esp32_nat_router has a Packet Capturing feature like the esp_wifi_repeater has?

spaceman5, 2022-02-03:
Hi Martin

The esp32_nat_router and the esp_wifi_repeater projects are terrific,
thank you very much for creating them.

I went over the description here for the esp32_nat_router,
and there was no mention there regarding a Packet Capturing feature..
Does it exist here too (and just was not mentioned), or it does not exist, and we can only get it on the ESP8266 NAT?

Thank you very much

martin-ger, 2022-02-04:
This feature is not implemented here. Be aware that the codebase between the ESP8266 and the ESP32 is completely different due to the different development environments.

spaceman5, 2022-02-04:
Oh, I see.

May I please ask you If it is something that will be implemented here in the future?
Or is it in the **NOT To Do** list?

Also I am curious to ask:
**>the codebase between the ESP8266 and the ESP32 is completely different due to the different development environments.**
I have not delved yet into the ESP32,
but I know that both the ESP8266and the ESP32, are programmed in C,
so the big difference between them is the Libraries?
they are completely different, and that's what makes uniting the code between the 2 impossible?

spaceman5, 2022-02-04:
I understand,
thank you very much for this, what a great and clear explanation.
I will stick to the ESP8266 project then, since I will use the Packet Capturing.

All the best to you for these cool projects.
Simply amazing what a small microcontroller can do..

closed 2022-02-04 by spaceman5