martin-ger/esp32_nat_router #81: Routing question

Freeejack, 2022-03-23:
Hi!

Great work, i really like this firmware but i have a question.

I tried to use this ESP wifi repeater method to extend my wifi to a greater range on location.

The situation is: 
(Sound mixer unit connected to main router via cable) -> (Wifi Router with 4G internet in a truck) - > (ESP32 with your firmware outside of the truck few meters away) <- (Tablet connected to ESP)

The firmware works great, i have internet on the tablet so it's fantastic, but:

The mixer IP is 10.20.1.56
Main router IP is 10.20.1.51
But the ESP IP is 192.168.x.x and the app on the tablet cannot find the mixer.

Can i setup the above config so that the tablet can reach the sound mixer?

Thank you very much if you can help, or any advice to solve the situation if it's not possible with ESP!

martin-ger, 2022-03-23:
The tablet should also have a 192.168.x.x IP, right? It should be able to ping the router and the mixer. Any device in the 10.20.1.0 network should see the ping coming from the tablet  some 10.20.1.x address, the upstream side of the ESP. Can you confirm this?

If the mixer tries to setup a connection back to the tablet, this should fail because of NAT. However, if you know the port number, you might try a portmap.

Freeejack, 2022-03-23:
First of all, thank you very much for your quick reply!

I tried to put things together this afternoon, but I didn’t succeed.

Of course, the tablet also gets an IP address of 192.168.4.2 or similar from ESP32 whose IP address is 192.168.4.1.
Tomorrow i can try the PING you mentioned, and i check back to reply on this!

It would be basically two-way communication. An application running on a tablet would control the mixer in some cases (check audio lines before work).
How can i setup a portmap to achive this? I need to set it up in the main router or on the ESP32, or both?

Somehow I find out which port the mixer is communicating with, there must be some information about it.

Thanks again for your help!