martin-ger/esp32_nat_router #84: Can not se devices on my main network

denko81, 2022-04-18:
I installed this and connected one esp cam but I can't find the cam. I can however use the wifi on my phone. Any ideas?

J-Rios, 2022-06-29:
Have you install esp32_nat_router firmware on an esp-cam device and you expect to also get "camera" functionality?

Or you mean that you have installed the esp32_nat_router on another esp32 device and then connect the esp-cam to that device AP and you are not able to get/know the IP address that is given to the esp-cam?

denko81, 2022-07-01:
The second option.

J-Rios, 2022-07-02:
Note then that the NAT router creates a subnet in 192.168.4.X and the IP addresses provided through DHCP to devices that connect to this NAT will be in that range, so you need to be able to reach that range from any other device from where you want to access the esp-cam.

Also, if you want to know which IP address has been added to the esp-cam, I think it could be obtained from esp-cam serial port, otherwise you can use an IP address network analyzer software like:
- [Linux] nmap
- [Windows] Advanced IP Scanner
- [Android] Fing
- ...