martin-ger/esp32_nat_router #107: [Question] Connected device not showinng up on upstream network

Aephir, 2022-09-17:
I'm trying to use this as a range externder for my IoT network. Just for one single device (an OpenEVSE car charger that's outside a brick wall). 

I've connected the ESP32 to my IoT network, can access it on it's new IP. I can see that the OpenEVSE connects to it (by the OpenEVSE AP disappearing, and the blinking pattern of the ESP32 indicating one device connected).

But I can't see the OpenEVSE anywhere upstream, it never appears on the network, so I can't connect to it. Any idea what I'm doing wrong?

I've tried both with `AP Settings` empty, and by copying the SSID and password from the `STA Settings`.

flix-one, 2022-11-02:
The problem is that you're car charger is in it's own subnet "192.168.4.0/24" and therefore you can only see up until the ESP32 with it's own IP in your main net (e.g. 192.168.0.0/24 or similar)
I've got the same problem with a smart relay (Shelly 1) that has a webserver on port 80 (in my case 192.168.4.2:80)

What I tried today was doing a portmap --> 192.168.0.141:5001 to 192.168.4.2:80

portmap [add|del] [TCP|UDP] <ext_portno> <int_ip> <int_portno>

Procedure :
Connect the ESP32 via UART to your PC and open the terminal, for example in Arduino IDE 
then enter task: portmap add TCP 5001 192.168.4.2 80

When entering: show
it should show the portmap

Unfortunately this didn't work for me and I can't understand why.
Are there additional steps to be taken before the portmap is active and working?
@martin-ger Maybe you can help here?

HarrysLapTimer, 2023-01-14:
I'm facing the exact same problem. Shelly (192.168.4.2) connected to the ESP_nat_router which is connected to my IoT VLAN with IP 192.168.3.115. I can ping 192.168.3.115. I can connect to the Shelly using its own AP to verify it is connected to ESP_nat_router.

I have added a static route for 192.168.4.0/24 via 192.168.3.115 on my main router. After having done this, I'm able to ping the Shelly from the IoT VLAN: 192.168.3.* ‣ 192.168.3.115 ‣ 192.168.4.2. However, I'm not able to connect to port 80 of the Shelly using a browser or curl.

I will try the port forwarding too, but I understand this didn't work for @flix-one.

Btw. I have enabled MQTT communication for the Shelly with a MQTT server on the IoT VLAN. The Shelly both publishes topics and receives notifications on subscribed topics. So this outbound communication is working fine.

martin-ger, 2023-01-14:
It is a NAT router, meaning that all devices in the APs network (usually the 192.168.4.0/24 network) will be hidden behind the STA side adress of the ESP. They can all connect to the outside and usually the Internet, but reverse connections (pinga, portscans etc.) will not succeed.

The only way to establish a portmap (port forward) from a port of the ESP on the STA side adress to a port of an internal device.

If you want to access e.g. a web server on 192.168.4.2, use:
`portmap add TCP 81 192.168.4.2 80`

Now you can see the server port 80 of the internal device externally (unter the adress of ESP's STA side) as port 81.

HarrysLapTimer, 2023-01-15:
Thank you @martin-ger for the clarification. I will check the port mapping for the 80 port. For pinging - the work around is to define a static route in the connecting subnet. The other option is the named MQTT communication. Due to the publish / subscribe mechanism, there is only outbound connection making - which is working nicely.