martin-ger/esp32_nat_router #112: portmapping does not get restored effectively after restart

mklingn, 2022-11-13:
Thanks for this great piece of work!

I have a single client behind my esp32_nat_router. It is an esp32-cam with esphome firmware and a fixed IP of 192.168.4.121

Via serial console I do a port mapping to connect to the esp32-cam's port 6053:

portmap add TCP 6054 192.168.4.121 6053 

This works, until the next restart of esp32_nat_router. Although the esp32-cam gets re-connected  successfully, connections to the forwarded port do time out.

"SHOW" still lists the portmapping, but it just does not seem to get applied - I looked at the code and it SHOULD be applied... Things I tried:

- wait for an hour
- restart the esp32-cam

Connectivity via portmapping can only be restored by adding another, identical port mapping via the console, or removing and re-adding the existing one.

I suppose I have a strange combination here, because persistent port mapping does work for others, right?

geidien, 2023-10-24:
Did you ever get any resolution to this issue?

mklingn, 2023-10-24:
IIRC the solution was to use https://github.com/martin-ger/esp_wifi_repeater with an esp8266 :-/