martin-ger/esp32_nat_router #118: power management and nat control

sgrizzi, 2023-01-03:
Hi, I have been using the esp32 router for a while. It works but sometimes it does generate a very strong output signal which interferes with the wifi interfaces of nearby devices - namely a wifi printer.
Is there any possibility to reduce the esp32 wifi output level? using this for example? `esp_wifi_set_max_tx_power(8);  `

I am also using the esp32 router as a tester system to simulate a "poor wifi connection" and a "remote server down" situation, so that I can test if wifi code is sufficiently robust. Is there any way to manually stop/restart the nat process? Could you recommend a way to do this?
Rgds

martin-ger, 2023-01-03:
Haven't any experience with esp_wifi_set_max_tx_power() - just try it! It will most probably have influence on the STA as well as on the AP.

You can disable NAPT using the lwip function "ip_napt_enable(u32_t addr, int enable)" for your interface.

sgrizzi, 2023-01-03:
Hi, thanks for the quick response.
As a quick&dirty solution, and just for testing, I embedded these lines at the end of the “show” command:
```
    int8_t power;
    esp_err_t power_error = esp_wifi_get_max_tx_power(&power);
    printf("output error % d power %d\n\r", power_error, power );

    power_error = esp_wifi_set_max_tx_power(8);
    printf("set output error % d\n\r", power_error );

    power_error = esp_wifi_get_max_tx_power(&power);
    printf("output2 error % d power %d\n\r", power_error, power );
```
This shows the current power level = 80 after initialization and sets it to 8 after command execution.
Power will be displayed set at 8 at a subsequent execution.

It works. The output power is indeed reduced – both for STA and AP.

I will try with ip_napt_enable().

For both cases I need to find a nice way to embed them either in a dedicated command, or, better, in the GUI interface.
Any recommendation there?

Best regards.

martin-ger, 2023-01-03:
You can add CLI commands in cmd_router.c (see the other examples).

The GUI is defined in pages.h and processed in http_server.c

sgrizzi, 2023-01-03:
Thanks a lot !

closed 2023-01-03 by sgrizzi