martin-ger/esp32_nat_router #119: show command always displays 0 stations connected

sgrizzi, 2023-01-04:
Hi, maybe it's me doing something wrong, but the "show" command is always displaying 0 stations connected, and the led is also never blinking....
At startup I get the following messages:
```
␛[0;32mI (3790) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2␛[0m
W (4070) wifi:<ba-add>idx:4 (ifx:1, 34:94:54:b7:20:f4), tid:0, ssn:0, winSize:64
␛[0;32mI (4580) esp_netif_handlers: sta ip: 192.168.1.131, mask: 255.255.255.0, gw: 192.168.1.1␛[0m
␛[0;32mI (4580) ESP32 NAT router: got ip:192.168.1.131␛[0m
␛[0;32mI (4590) ESP32 NAT router: set dns to:192.168.1.1␛[0m
```
and the device "192.168.4.2" is actually connected to the router and working fine. The led (mapped to GPIO 25) is not blinking.

The "show" command displays the following:
```
show
␛[0;32mI (31280) cmd_router: ssid **********␛[0m
␛[0;32mI (31280) cmd_router: passwd **********␛[0m
␛[0;32mI (31280) cmd_router: ap_ssid **********␛[0m
␛[0;32mI (31280) cmd_router: ap_passwd **********␛[0m
STA SSID: ********** Password: ********** Enterprise: <not active> 
AP SSID: ********** Password: **********
AP IP address: 192.168.4.1
Uplink AP not connected
0 Stations connected
```
I would be expecting to see at least 1 station conncted and the led to blink accordingly.
Am I wrong? or am I doing something wrong?
Rgds.

martin-ger, 2023-01-04:
That's wrong: given the log before is correct, it should be "Uplink AP connected" and "1 Stations connected".

My test env does it like this...

sgrizzi, 2023-01-04:
Clear - it was also my expectation.
I just compiled the original code in Platformio - the only change is the mapping of the onboard LED. The code compiles with no problem... and works also fine.
I will need to investigate further.

sgrizzi, 2023-01-04:
Maybe I found the problem...
In the wifi_event_handler, WIFI_EVENT_AP_STACONNECTED and WIFI_EVENT_AP_STADISCONNECTED are WIFI_EVENTs i.o. IP_EVENTs, I guess.
Replacing these lines around line# 346 in es32_nat_router.c apparently solves the problem.

```
    //else if (event_base == IP_EVENT && event_id == WIFI_EVENT_AP_STACONNECTED)
    else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_AP_STACONNECTED)

    //else if (event_base == IP_EVENT && event_id == WIFI_EVENT_AP_STADISCONNECTED)
    else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_AP_STADISCONNECTED)
```

Station count is now correct and the LED is also blinking!
Still, Uplink AP is displayed as not connected.

sgrizzi, 2023-01-04:
Is there a data structure where I can find the IPs of the currently connected stations? any "easy" way to retrieve this information?

martin-ger, 2023-01-04:
I am a little bit confused: in the current codebase these are WIFI_EVENTs ( https://github.com/martin-ger/esp32_nat_router/blob/master/main/esp32_nat_router.c#L348 ), aren't they? Do you use the current main branch?

sgrizzi, 2023-01-04:
Apologies. You are right. I downloaded the zip file on Oct.28th and the new version is of Nov.9th... and there events are indeed WIFI_EVENTS.
I will replace my older files...

Can I still get a hint from you on this, please? "Is there a data structure where I can find the IPs of the currently connected stations? any "easy" way to retrieve this information?"

martin-ger, 2023-01-04:
Guess you have to trace this info collecting the connected MACs from struct wifi_event_ap_staconnected_t in WIFI_EVENT_AP_STACONNECTED and then later requesting their IPs from the DHCP server using esp_netif_dhcps_get_clients_by_mac(). Probably not easy, but possible...

sgrizzi, 2023-01-04:
OK! Thanks a lot!

Meanwhile, the new files also fix the Uplink connected status... but the displayed IP is always 0.0.0.0
Is this correct?

```STA SSID: ****** Password: ****** Enterprise: <not active> 
AP SSID: ****** Password: ******
AP IP address: 192.168.4.1
Uplink AP connected
IP: 0.0.0.0```

martin-ger, 2023-01-05:
Not okay - think fixed it with the lastet push.

closed 2023-01-05 by martin-ger

sgrizzi, 2023-01-05:
Hi Martin, the fix is working perfectly.

I added the following code at the end of the "show" command to display the list of connected stations, which I find useful:
```
    //get list of stations conncetd to the AP
    wifi_sta_list_t sta;    
    esp_wifi_ap_get_sta_list(&sta);
    for (uint8_t jj=0; jj<sta.num; jj++){
    printf("mac addr: " MACSTR "\n", MAC2STR(sta.sta[jj].mac));
```
Pretty straightforward and I get the following output:
```
STA SSID: ******* Password: ******* Enterprise: <not active> 
AP SSID: ******* Password: *******
AP IP address: 192.168.4.1
Uplink AP connected
IP: 192.168.1.131
2 Stations connected
mac addr: 34:94:54:b7:20:f4
mac addr: 52:11:46:d0:3c:03
```
Unfortunately I could not manage to get the related IP addresses.
Apparently the function esp_netif_dhcps_get_clients_by_mac() does not exist in my esp_netif.h file, while this function does exist in esp_netif.h visible on github.
I upgraded esp32 platform on Platformio to version 5.3.0, cleaned and recompiled... still that function is not appearing...
I know it is not related to your code, but would you have any suggestion on how to solve the issue? or any alternative way to get the IP addresses?
Rgds