martin-ger/esp32_nat_router #120: unable to retrieve IPs assigned by AP to connected stations

sgrizzi, 2023-01-06:
Hi Martin, I am referring to previous issues #118 and #119 and your suggestions
>You can disable NAPT using the lwip function "ip_napt_enable(u32_t addr, int enable)" for your interface.

I did manage to list all mac's of the connected stations using this code
```
//get list of stations connected to the AP
    wifi_sta_list_t sta;    
    esp_wifi_ap_get_sta_list(&sta);
    for (uint8_t jj=0; jj<sta.num; jj++){
    printf("mac addr: " MACSTR "\n", MAC2STR(sta.sta[jj].mac));
```
but I cannot get the IPs related to each mac.
>...then later requesting their IPs from the DHCP server using esp_netif_dhcps_get_clients_by_mac(). Probably not easy, but possible...

Apparently the function esp_netif_dhcps_get_clients_by_mac() does not exist in my esp_netif.h file, while this function does exist in esp_netif.h visible on github.
The same is true for function "ip_napt_enable(u32_t addr, int enable)". It looks "empty". Even if I try and enter the relevant parameters manually, nothing happens.... (nor do I get any error...)

I am compiling under Platformio, esp platform 5.3.0, and I get this info when I compile
```
 - framework-espidf @ 3.40403.0 (4.4.3) 
 - tool-cmake @ 3.16.4 
 - tool-esptoolpy @ 1.40400.0 (4.4.0) 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
```
For some reasons my components library looks not as complete/updated as the one on github.
Would you have any idea on how to solve/bypass this issue?
Rgds.
(btw, thanks for your support. I managed to register a new CLI command to control wifi output power. I can share the code if of any interest)

sgrizzi, 2023-01-07:
Hi, I did manage to retrieve IP info using this function:
dhcp_search_ip_on_mac(u8_t *mac, ip4_addr_t *ip);
Here is the new code:
```
    //get list of stations connected to the AP
    wifi_sta_list_t sta;    
    esp_wifi_ap_get_sta_list(&sta);
    for (uint8_t jj=0; jj<sta.num; jj++){
    printf("mac addr: " MACSTR " - ", MAC2STR(sta.sta[jj].mac));
        ip4_addr_t ip;
        if (dhcp_search_ip_on_mac(sta.sta[jj].mac, &ip)) {
            // IP address was found for the MAC address
            printf("IP addr: " IPSTR "\n", IP2STR(&ip));
        }
    }
```

Unfortunately this function still does not work: "ip_napt_enable(u32_t addr, int enable)". It looks "empty". Even if I try and enter the relevant parameters manually, nothing happens.... (nor do I get any error...)
... any suggestions?

martin-ger, 2023-01-07:
Thank you for your feature enhancements. I will include that in the next push.

The code of ip_napt_enable() is certainly not empty (otherwise the whole SW should not work at all). I can find it in esp-idf/components/lwip/lwip/src/core/ipv4/ip4_napt.c
```
void ip_napt_enable(u32_t addr, int enable)
{
  struct netif *netif;
  int napt_in_any_netif = 0;
  for (netif = netif_list; netif; netif = netif->next) {
    if (netif->napt)
      napt_in_any_netif = 1;
    if (netif_is_up(netif) && !ip_addr_isany(&netif->ip_addr) && netif->ip_addr.u_addr.ip4.addr == addr && enable) {
      netif->napt = 1;
      ip_napt_init(IP_NAPT_MAX, IP_PORTMAP_MAX);
      break;
    }
  }
  if (!enable && !napt_in_any_netif) {
    for (netif = netif_list; netif; netif = netif->next)
      netif->napt = 0;
    ip_napt_deinit();
  }
}```

sgrizzi, 2023-01-07:
You are most welcome. I will also share the code to control output power... when it is solid.

Yes, you are right. I did look into /.platformio/packages/framework-espidf/components/lwip/lwip/src/core/ipv4/ip4_napt.c
and the code is indeed there!

I tested the function "hard-coding" the ip address of the station I want to disable (192.168.4.4) by calling it in this way:
`ip_napt_enable(0x0404A8C0, 0)` while the station is connected to the router.
Nothing is visible on console output.
The station (actually an iPhone) remains connected to the router -fine- and can still navigate on the internet as before, i.e. connect to the web via the esp-router -which is not was I was expecting.
Either I am doing something wrong, or is my expectation wrong ?

sgrizzi, 2023-01-07:
Clear, then my expectation was wrong.
>What exactly do you want to achieve with the call?

Actually I would like to disable the possibility of a station to connect to the web, while retaining its connection to the router.
Basically I would like to simulate a "remote server down" situation.
As an alternative, also disabling the router STA access altogether would do, but I would like to retain the connections active on softAP. 

I asked _chatGTP_ (the AI engine 😁 ) for suggestions, and it was stating that I could do it like this 
`nat static_rule add --src_mac AA:BB:CC:DD:EE:FF --src_ip 192.168.4.5 --allow false`
from command line in your code... but I think the engine was actually looking into the future as I don't see this command implemented as yet, right...?

gjroots, 2023-05-09:
HI. @martin-ger, @sgrizzi 
  
Try this code. https://github.com/gjroots/esp32_nat_router_plus/blob/main/components/wifi_handler/wifi_handler.c#LL121C1-L130C6

This is simple, I made like this. but call after some time dhcp assigned IP to clients other wise it will show 0.0.0.0 due to dhcp still not assign IP to clients.