martin-ger/esp32_nat_router #125: Suggestion: Add more features to Web Interface

OwensDataCenter, 2023-02-14:
I suggest that we add the ability to forward ports via the web interface and have the ability to scan for Wi-Fi networks via the serial console and web interface. Because the router could be remote and there is probably no way to use the serial console.

I have some knowledge of ESP-IDF and have compiled this firmware many times, All I need is some code to work with and I will probably get it working.

FrankNT, 2024-07-14:
Hello, I've added portmap feature (forward ports) to the web interface
I've created a pull request here: https://github.com/martin-ger/esp32_nat_router/pull/167
Feel free to help me on this feature, I'm don't have much experience in C programing :D

I committed my code on this https://github.com/FrankNT/esp32_nat_router , you can check here in case my pull request is not accepted :)

Screenshot:

[image]

damienwolf07, 2024-07-14:
@phucnguyentrong, Got it!
I have created a pull request on your fork, Try it and see if it works, I will await your feedback ;3

phucnguyentrong, 2024-07-17:
@damienwolf07 Thanks, bro. It works like a charm
[image]

But I have to change some lines of code, from "auto portmethodOK" to "int portmethodOK", etc... Since my compiler seems not understand "auto".
Plus, please don't format code that you don't modify, because it may confuse the reviewer.

Thanks again :)

damienwolf07, 2024-07-17:
> But I have to change some lines of code, from "auto portmethodOK" to "int portmethodOK", etc... Since my compiler seems not understand "auto".
Sorry, I have a fix for that, I didn't have the time for looking what type the parameters were.

> Plus, please don't format code that you don't modify, because it may confuse the reviewer.
Yeah, but there is a reason why, the lines were all over the place and it was bothering me.