martin-ger/esp32_nat_router #128: VS Code+PlatformIO + Espressif IDF build errors

apis-makerkang, 2023-04-22:
Hi, first, I would like to thank you for making this esp_nat_router supporting WPA2 Enterprise. 
I tried to build it under the environment below
- Windows 10
- VS Code 
- PlatformIO Extension (Core: 6.1.6, Home:3.4.3)
- Espreswif IDF Extension V1.6.1

I git clone from this repository,  use [env:esp32dev] to build, and get errors as
```
D:\temp\esp32_nat_router>C:\Users\paulchainkang\.platformio\penv\Scripts\platformio.exe run --environment esp32dev
Warning! Ignore unknown configuration option `monitor_flags` in section [env:ttgo-lora32-v1]
Warning! Ignore unknown configuration option `monitor_flags` in section [env:esp32dev]
Warning! Ignore unknown configuration option `monitor_flags` in section [env:esp32-c3-devkitm-1]
Processing esp32dev (platform: espressif32; framework: espidf; board: esp32dev)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50001.0 (5.0.1)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40500.0 (4.5.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-xtensa-esp32 @ 11.2.0+2022r1
Reading CMake configuration...
Generating assembly for certificate bundle...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\esp32dev\main\esp32_nat_router.o
Compiling .pio\build\esp32dev\main\http_server.o
Generating LD script .pio\build\esp32dev\memory.ld
Compiling .pio\build\esp32dev\app_trace\app_trace.o
Compiling .pio\build\esp32dev\app_trace\app_trace_util.o
Compiling .pio\build\esp32dev\app_trace\host_file_io.o
Compiling .pio\build\esp32dev\app_trace\gcov\gcov_rtio.o
Compiling .pio\build\esp32dev\app_trace\port\port_uart.o
In file included from main/http_server.c:23:
main/pages.h:167:70: warning: backslash-newline at end of file
  167 | #define LOCK_PAGE "HTTP/1.0 200 OK\r\nContent-Type: text/html\r\n\r\n\
      |
In file included from main/http_server.c:24:
components/cmd_router/router_globals.h:43:23: error: unknown type name 'u8_t'
   43 | esp_err_t add_portmap(u8_t proto, u16_t mport, u32_t daddr, u16_t dport);
      |                       ^~~~
components/cmd_router/router_globals.h:43:35: error: unknown type name 'u16_t'
   43 | esp_err_t add_portmap(u8_t proto, u16_t mport, u32_t daddr, u16_t dport);
...
  469 |     dnsserver.type = IPADDR_TYPE_V4;
      |              ^
main/esp32_nat_router.c:469:22: error: 'IPADDR_TYPE_V4' undeclared (first use in this function); did you mean 'ESP_IPADDR_TYPE_V4'?
  469 |     dnsserver.type = IPADDR_TYPE_V4;
      |                      ^~~~~~~~~~~~~~
      |                      ESP_IPADDR_TYPE_V4
cc1.exe: some warnings being treated as errors
*** [.pio\build\esp32dev\main\esp32_nat_router.o] Error 1
============================================= [FAILED] Took 27.82 seconds =============================================

Environment    Status    Duration
-------------  --------  ------------
esp32dev       FAILED    00:00:27.824
======================================== 1 failed, 0 succeeded in 00:00:27.824 ========================================

D:\temp\esp32_nat_router>
```
I googled and find someone said changing "platform = espressif32" to "platform = espressif32@5.3.0" may help. I modified 
[env:esp32dev] to
```
[env:esp32dev]
platform = espressif32@5.3.0
framework = espidf
board = esp32dev

monitor_speed = 115200
monitor_flags = --raw
```
Then build and get error of "Error: Could not find one of 'package.json' manifest files in the package"
```
D:\temp\esp32_nat_router>C:\Users\paulchainkang\.platformio\penv\Scripts\platformio.exe run --environment esp32dev
Warning! Ignore unknown configuration option `monitor_flags` in section [env:ttgo-lora32-v1]
Warning! Ignore unknown configuration option `monitor_flags` in section [env:esp32dev]
Warning! Ignore unknown configuration option `monitor_flags` in section [env:esp32-c3-devkitm-1]
Processing esp32dev (platform: espressif32@5.3.0; framework: espidf; board: esp32dev)
------------------------------------------------------------------------------------------------------------------------
Error: Could not find one of 'package.json' manifest files in the package
```
Please someone help.  Many thanks in advance.

gjroots, 2023-05-09:
Hi..  @apis-makerkang , may be downgrade from framework-espidf @ 3.50001.0 (5.0.1) to 4.x.x version may works for you because i am  not facing any issue while using  framework-espidf @ 3.40401.0 (4.4.1) 

this is i am currently using version it works fine for me :

- framework-espidf @ 3.40401.0 (4.4.1)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.22851.191205 (2.28.51)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3

apis-makerkang, 2023-05-14:
@gjroots Thank you for providing helps. I tried to change the environment in platformio.ini with espidf @ 3.40401.0 like
```
[env:esp32dev]
platform = espressif32
framework = espidf @ 3.40401.0
board = esp32dev
```
But I got error of "Error: This board doesn't support espidf @ 3.40401.0 framework!",
```
Executing task: C:\Users\paulchainkang\.platformio\penv\Scripts\platformio.exe run --environment esp32dev 

Processing esp32dev (platform: espressif32; framework: espidf @ 3.40401.0; board: esp32dev)
----------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.1.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
Error: This board doesn't support espidf @ 3.40401.0 framework!
 - tool-esptoolpy @ 1.40500.0 (4.5.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
======================================= [FAILED] Took 0.93 seconds =======================================
```
How can I downgrade to framework-espidf @ 3.40401.0 (4.4.1)?

gjroots, 2023-05-14:
Hi @apis-makerkang,
 I wanted to point out that this project supports up to espidf 4.x.x, and with the release of espidf 5.x.x, some breaking changes have occurred. To resolve this issue, you can change the Espressif 32 version in the  `platformio > platforms > embedded menu`  to version 5.1.0. This should work, and if you want to check the change log, version 5.1.0 to 5.3.0 have espidf 4.4.1 version or 4.4.2 as you wish. Currently, I am using version 5.1.0.  if you carefully watch the log currently you are in `PLATFORM: Espressif 32 (6.1.0) > Espressif ESP32 Dev Module` . platfrom version  and espidf version are different in platformio

let me see is it work for you or not!.

[image]

apis-makerkang, 2023-05-29:
Hi @gjroots , sorry the the late response. I tried your last suggestion to set  platformio.ini with
[env:esp32dev]
platform = espressif32 @5.1.0
framework = espidf
board = esp32dev

Then it compiles OK without problem. Thank you very much.

closed 2023-05-29 by apis-makerkang