martin-ger/esp32_nat_router #131: Power consumption while connected to upstream AP but with no clients connected

Istria1704, 2023-05-27:
First of all thank you very much for making this available to us all. Very useful software. 

I want to make a solar/battery powered repeater for my garden. But to design the solar panel and battery size, I need to know the average power consumption when idle. "Idle" in this case meaning "connected to upstream AP but no clients connected". 
I see the 1,8W power when at top speed, which would correspond to almost 500mA drawn from a 3.7V lipo. But if I use that, I'd need a large battery to last through a long winter night. I assume it draws (much) less when not throughputting 15Mbit. Had anyone measured how much? 

For context: I plan to use it for 2 things:
1. To give wifi access to some esp8266 sensors in the garden. Those will not be connected and transmitting continuously, but sleep for 10-30min and only briefly wake up, connect, update sensor data, and go back to sleep. So 10-30min not connected. And 10s connected. 
2. Occasionally give wifi access to my phone and watch YouTube when sitting in the garden. So I will calculate using the 15Mbit/1.8W for this usage. 

Thanks in advance!

martin-ger, 2023-05-27:
I think, 70mA is a good guess for an ESP doing nothing but beeing connected 
to WiFi. Nothing you want to run for days on battery... :-(

Istria1704, 2023-05-27:
Oh, that is very reasonable. Even less than an esp8266. My Wemos D1 Mini pulls 70mA @ 5V. We are talking about ESP32, right? 

Do you mean 70mA at 3.3V, or at the (3.7v) battery, or at (5V) usb input? 

It won't have to last for days. In theory just 1 night. During the day, the battery will be charged by a small solar panel.

martin-ger, 2023-05-28:
At 3,3V - Try to avoid the 5V-3.V USB converter, this will consume at least 1/3 of the power. On a 600mA LiPo a naked ESP8266 runs about 6h in WiFi-repeater-mode.

Istria1704, 2023-05-28:
Thanks. 
I must say my initial crude "ballpark" power calculation had a critical flaw. I didn't incorporate that during winter, the solar panel yield is MUCH MUCH less than in summer.
For an average of 70 mA @ 3.3v, in January I would need a 6.5Wp solar panel to break even with the power consumption. So make that 10Wp panel to have some breathing room. And that already becomes a fairly large 25x20cm panel. And even then I would probably end up with an empty battery now and then when it's heavily overcast for a couple days in a row.

I'm starting to think maybe solar/battery isn't going to be the way to go.Maybe I'll place the repeater inside the house, on the garden window. That way I can just plug it into a wall socket.

But that's anyway for the 70mA (231 mW) average idle power consumption! I couldn't find that data anywhere in the web.