martin-ger/esp32_nat_router #134: Very slow speed

Cancro29, 2023-07-27:
I remember trying this firmware a few months ago and can get up to 15mbps, but now i barely reach 3mbps with average speed about 1mbps. My board is ESP-32 Devkit v1

lhtien, 2023-11-25:
Yes! Me too. It is too slow...
This is the result from the speedtest website.

[image]

piotrcurious, 2023-12-08:
Same here. i can barely reach 9mbit. 
I guess the precompiled binary distributed is optimized with Os ... perhaps one needs to build from source and optimize with Og/O3

MichelangeloGuaitolini, 2024-03-13:
Any development here? I am facing the same issue at the moment, being able to barely reach 6 Mbps

geidien, 2024-03-20:
Bumping this. Same issues for me. Not sure why there is a sudden drop in performance.

Cancro29, 2024-03-20:
I've ported the code into Arduino and tested various boards. Looks like the boards also affect speed. Here is the boards that i've tested: 
1. ESP32 DevKit V1 = ~3 mbps
2. Wemos ESP32 = ~9 mbps
3. Seeed Studio ESP32C3 = ~15 mbps

Still figuring out the answer of my question though. For now i can only assume that the board i've been using is a bad board

geidien, 2024-03-20:
I found the solution to my particular case. It turns out ISP's like to recommend and use their own DNS services these days. Once I updated my routers to Cloudflare DNS 1.1.1.1 and 1.0.0.1 everything was resolved.

Don't know if this is your exact problem, but it is a possibility!! Also check out esp32_nat_router_plus those guys are actively improving this project.