martin-ger/esp32_nat_router #150: I can't connect to esp32c3

Yodinsn, 2024-01-13:
When I use my "windows" computer to connect to my esp32c3 wifi, my computer said "can't not connect to this wifi".
But when I used my "android" phone  to connect the same wifi, it worked smoothly.
How do I setup esp32c3 or my computer.
Orz

nissimzur, 2024-01-13:
Hi
Is your esp33 configured as route?
Is your pc is laptop? Or desktop pc using rg45 cabel?
If is uses cable, it can not connect to any wifi
-Nissim

Yodinsn, 2024-01-13:
my desktop didn't any rj45 and it's icon will turn this one
[image]
but...it's next by my pc...

nissimzur, 2024-01-13:
Can you wifi scan and see all wifi networks around you (ssid names)?
If not, you pc wifi is locked.
We can whatsup voice talk, if you like. +972544500635. Nissim zur

Yodinsn, 2024-01-14:
I use esp8266 and esp_wifi_repeater now. 
And i want to upgrade to esp32c3(it's cheaper than esp32) and this project.

I can't use only this wifi...
The others is all good to use...
By the way...My english is very bad....

Yodinsn, 2024-01-17:
I'm changing to esp32 lite
But... still...