martin-ger/esp32_nat_router #154: NO XMC flash  detected!

Kuba1606, 2024-03-01:
Hi, i'd love to run Your project on my SpeedStudio XIAO ESP32 S3 but i have some issue with download tool on Windows. Maybe You can help me with it. As soon as I click on SART button cmd shows up this information: "NO XMC flash  detected!" but ignored it and continue his job until it's all finished with exit status:  "is stub and send flash finish". So I understand everything is OK. My board works fine with other sketches in Arduino IDE. My Wemos D1 mini ESP8266 also works fine with Download Tool on my PC. 
[image]
[image]

12Genna12, 2025-01-14:
hi, phyton+cmd fix this trouble))))

[image]

SWDevStudio, 2025-01-27:
I have the same problem

[image]

12Genna12, 2025-02-01:
Try dowload phyton + pip for python and flash via cmd. With the help of the official program, it did not work