martin-ger/esp32_nat_router #155: No internet connection from nat router

CoolmdXi, 2024-03-12:
I have follewd various examples of flashing esp32 with this software and everything appears to be same as examples but the router always shows as having no internet connection when connected to my own wifi. unable to ever get it to work. connected and flashed 3 different working esp32 but all showed and wifis with no internet connection. any advise would be welcome

imoldovavan, 2024-03-13:
I have the same problem with the latest firmware. This also happened before I flashed the latest firmware, so it may not be the esp32. I will try it on a different wifi network and let you guys know.

imoldovavan, 2024-03-13:
Installed it on a different board, a ttgo board and it works with no issues. I guess that one board that I have is bad somehow.

CoolmdXi, 2024-03-13:
Weird wish I had more luck tried on 3 esp32 boards all work perfectly with sketches from Arduino but all give the same result..connected no WiFi...oh well .. not much I can do. Glad you were lucky

rmarket6, 2024-03-31:
@CoolmdXi i ran into the same issue. I have the esp32-c3-devkitm-1U

essepunto, 2024-05-27:
same thing. But I have a WPA3 router, I'll try to switch to WPA2, then I'll write here.

Zodijackyl98, 2024-06-24:
Encountered with the same error, also tried to use ESP32 as a repeater using micropython firmware but could manage to get internet connection in that way neither.  Tried with my mobile phone's hotspot instead of router but result did not change. I don't quite understand what is the problem, maybe security level is the issue but I also tried it with my raspberrypi 4 which has lower security level 2 instead of 4 like common routers but again same issue still persists.

Zodijackyl98, 2024-06-24:
> Encountered with the same error, also tried to use ESP32 as a repeater using micropython firmware but could manage to get internet connection in that way neither. Tried with my mobile phone's [...]

I left my previous proxy settings active while trying to connect via my phone and since I used the same ssid that i configured my AP with Micropython, my phone automatically connected to the network with saved proxy settings which was wrong. Deleted the proxy settings, configured my network setting via serial console after flashing process and everything is okay now.