martin-ger/esp32_nat_router #162: ESP32S3. - Should this hardware work

candrews67, 2024-05-19:
I bought an ESP32-S3 for this but I have not been able to make it work.

The README.md has instructions for esp32 and esp32c but there is also a esp32s3 directory under build.    I have attempted to install the esp32s3  binaries using the provided esptool examples for esp32 and esp32c3 but they do not work.

Do I need to compile for esp32s3 or will this chip not work?

Thanks!!!

Ahsan-Himu, 2024-07-11:
> I bought an ESP32-S3 for this but I have not been able to make it work.
>
> The README.md has instructions for esp32 and esp32c but there is also a esp32s3 directory under build. I have attempted to [...]

I am also facing same problem with my ESP32s3 ,Please help if anyone know something regarding this.

sliacanm, 2024-08-03:
Hey there, I just tested ESP-S3 and flashed with no issues. However, I did build from scratch before flashing. These steps worked for me: 
cd to esp32_nat_router folder
Delete the build folder
get_idf or run the export.sh as recommended in the ESP-IDF install -->#e.g. for me on MAC. $HOME/esp/esp-idf/export.sh
cmf: idf.py set-target esp32s3
In the cmd:"idf.py menuconfig", make sure all the necessary settings are enabled as stated in https://github.com/martin-ger/esp32_nat_router#building-the-binaries
cmd: idf.py build
cmd: idf.py -p PORT flash #e.g. for me: idf.py -p /dev/cu.usbmodem01 flash

JanisPlayer, 2024-08-18:
@sliacanm Thanks, it worked.
Since creating the build is really time-consuming, I’m sharing my finished build [esp32s3.zip](https://github.com/user-attachments/files/16652442/esp32s3.zip).

nopnop2002, 2025-07-22:
>Is there anything else I need to know when flashing this?

You need to know in advance the size of the SPI FLASH your board has.

jimbeam128, 2025-08-03:
I also had problems to get my ESP32 S3 to work. When I flashed with the ESP flash download tool it did not work. 
I already changed the flash size to 4 mb in platformio.ini

[env:esp32-s3]
board = esp32s3box
board_upload.flash_size = 4MB

I finally got it to work when I flashed it via command line with ESP-IDF. I used Version 5.2.5 for compatibility reasons.

Here´s the command which did work for me:

python -m esptool --chip esp32s3 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 4MB --flash_freq 80m 0x0 build\bootloader\bootloader.bin 0x8000 build\partition_table\partition-table.bin 0x10000 build\esp32_nat_router.bin