martin-ger/esp32_nat_router #173: Using IP DHCP server from gateway

karnasw, 2024-09-10:
Hi, are there any way to use DHCP server from main router (gateway)? Actually devices connected to esp32 nat have nat 's IP subnet.

vks007, 2025-01-12:
I am also looking for this solution. My purpose is to use this extender for IOT devices but they have to communicate back to the server on the main router's network. ESP32 NAT creates its own subnet and clients connected to it arent accessible from the main router's network. Am I missing something here?

georgepapas123, 2025-02-07:
@karnasw To achieve this you have to use the ```portmap``` command from a serial terminal (MobaXterm is one of the best).
The following is an excerpt from README.md:
 
```
Access devices behind the router

If you want to access a device behind the esp32 NAT router? PC -> local router -> esp32NAT -> server

Lets say "server" is exposing a webserver on port 80 and you want to access that from your PC.
For that you need to configure a portmap (e.g. by connecting via the arduino IDE uart monitor through USB)

portmap add TCP 8080 192.168.4.2 80
                                 ↑ port of the webserver
                            ↑ server's ip in esp32NAT network
                  ↑ exposed port in the local router's network

Assuming the esp32NAT's ip address in your local router is 192.168.0.57 you can access the server 
by typing 192.168.0.57:8080 into your browser
```
In other words, if you typed ```192.168.0.57:80``` (or just ```192.168.0.57```) you would be accessing the Web Config Interface page (same as ```192.168.4.1``` from esp32NAT's network), but now traffic to port 8080 goes to your device so it is now also accessible as ```192.168.0.57:8080``` (same as ```192.168.4.2```)
If it helps think of it as 
```192.168.4.1``` = ```192.168.0.57```
```192.168.4.2``` = ```192.168.0.57:8080```

@vks007 If your device (```192.168.4.2```) serves a web page, the above will allow you to access it, but you won't be able to access the rest of ```192.168.0.x``` network from your device.