martin-ger/esp32_nat_router #176: Build for ESP32-S3 supermini - a solution

fiz1962, 2024-10-18:
The prebuilt binaries don't work for my ESP32-S3 supermini.  Nor does rebuilding with the default settings.  But changing the Flash size from 16MB to 4MB does work.  Thanks for the repo.

Striagor, 2024-11-07:
For information, this method also works on ESP32-s3-zero boards! thanks for the information!

RutishGN, 2025-04-18:
> The prebuilt binaries don't work for my ESP32-S3 supermini. Nor does rebuilding with the default settings. But changing the Flash size from 16MB to 4MB does work. Thanks for the repo.

How to change flash size ?

adarazs, 2025-05-31:
This tip helped me, thanks Striagor! Also had to research how to fix the flash size in `platformio.ini` so here, in case anybody else needs it, I added the last line:

```
[env:esp32-s3]
board = esp32s3box
board_upload.flash_size = 4MB
```

Which made it work on a Lolin S3 mini.

KORNELCS2, 2025-10-24:
@fiz1962 
I used esp idf and it works now.

But I don't know platformio and I tried that first. That's why I wrote it down. 

I found specific files in another issues tab and it works.